robotology-legacy/wysiwyd #10: [opcEars] Comparator does not work

Tobias-Fischer, 2015-02-06:
Hi all,

somehow the != comparator does not work when looping over beliefs in wrdac/opcEars. I did a quick fix in https://github.com/robotology/wysiwyd/commit/0756ec38d164a6d1abc7aabd7b2bd69b4c6a1328 to check for 
```size > 0```
But it should be done properly, otherwise someone else might run into this issue sooner or later.

Would be great if someone from WP5 could have a look at it :).

Thanks,
Tobias

pattacini, 2015-04-27:
Any update on this? Otherwise I think we can close it.

Tobias-Fischer, 2015-04-27:
I don't think it should be closed yet. The comparator still doesn't work, and it is just a matter of time until someone else runs into issues related with this. And nobody likes SegFaults, right?!? :)

pattacini, 2015-04-27:
Then we should assign to someone of **WP5**, otherwise it won't get solved ever :smirk:

pattacini, 2015-04-27:
@jypuigbo @clement-moulin-frier would you be willing to solve this?

gregoire-pointeau, 2015-04-27:
Hum, I can take a look at it this week if you want. Because I don't think
Jordi or Clement are used to OPCEARS.

pattacini, 2015-04-27:
Fine with me, thanks @gregoire-pointeau

Tobias-Fischer, 2015-06-25:
Fixed this with https://github.com/robotology/wysiwyd/commit/399e81a61466a10198bb816f56bd06cf8842c08f

closed 2015-06-25 by Tobias-Fischer