robotology-legacy/wysiwyd #11: rekognition fails to compile

pattacini, 2015-03-11:
@maxime-petit could you address this [problem](https://ci.appveyor.com/project/pattacini/wysiwyd/build/1.0.502) and the cmake warning reported below?

```
CMake Warning at conf/wysiwydFindDependencies.cmake:33 (find_package):
  By not providing "FindRTABMap.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "RTABMap", but
  CMake did not find one.

  Could not find a package configuration file provided by "RTABMap" with any
  of the following names:

    RTABMapConfig.cmake
    rtabmap-config.cmake

  Add the installation prefix of "RTABMap" to CMAKE_PREFIX_PATH or set
  "RTABMap_DIR" to a directory containing one of the above files.  If
  "RTABMap" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:48 (include)

CMake Warning at conf/wysiwydFindDependencies.cmake:34 (find_package):
  By not providing "FindPCL.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "PCL", but
  CMake did not find one.

  Could not find a package configuration file provided by "PCL" with any of
  the following names:

    PCLConfig.cmake
    pcl-config.cmake

  Add the installation prefix of "PCL" to CMAKE_PREFIX_PATH or set "PCL_DIR"
  to a directory containing one of the above files.  If "PCL" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:48 (include)
```

Tobias-Fischer, 2015-03-11:
Hi Ugo,

that was my fault, sorry. The actual problem was that CURL was not found on the Windows machine. Now, the build for rekognition is only triggered if CURL is found.
The AppVeyor build still fails, but because wget could not be installed. Can you please manually trigger another build?

Best, Tobias

pattacini, 2015-03-11:
Thanks @Tobias-Fischer 
However, I still have the annoying cmake warning.
I'll try to trigger Appveyor again but I'm afraid we're having general failure with `wget`. Possibly, I'm going to try to rely on a different downloading tool... stay tuned.

Tobias-Fischer, 2015-03-11:
Hi @pattacini 
Warnings should be gone now :). Hope you get the wget issue resolved. Maybe it was just temporarily down?

closed 2015-03-11 by pattacini