robotology-legacy/wysiwyd #12: [ci]: need for updated yarp packages

pattacini, 2015-04-23:
Currently `travis` and `appveyor` CI systems fail to compile `wysiwyd` project because `idl` auto-generated files contain references to the new `yError()` routine and thus `yarp` headers need to be updated.

`travis` and `appveyor` have been temporarily disabled: remember to enable them again, once new `yarp` package will be released.

lornat75, 2015-04-24:
@mbrunettini has been looking at this, I think he has been busy with compiling ipopt for visual studio 2013.

pattacini, 2015-04-24:
there's no hurry

pattacini, 2015-06-18:
Given the availability of the latest releases of `yarp` and `icub-main` I could resume the CI services again.

##### AppVeyor
We should have it functioning, aside from compilation problems that might arise due to specific `wysiwyd` issues.

##### Travis
Unfortunately, we have troubles finding `yarp` from within `cmake` after the installation of `icub` packages.
See https://travis-ci.org/robotology/wysiwyd/builds/67426446 for the report.
we have to inspect this issue.

/cc @mbrunettini @lornat75 @robotology/wysiwyd-developers

Tobias-Fischer, 2015-06-19:
Hi @pattacini,
Did you notice that there is no new version of `yarp` and `icub-main` for `precise`? `trusty` seems to be the minimum requirement for `yarp` 2.3.64 / `icub-main` 1.1.16, but the Travis build environment is `precise` ..

This makes the problem that `yarp` isn't found by `cmake` even more weird, as the old versions used to work previously?!?

Best, Tobias

pattacini, 2015-06-19:
yes, thanks @Tobias-Fischer, problem spotted, we are working on it.
I think I'll give soon a try to:
- `icub-common` via `apt-get` (we use the old one, `ubuntu/precise`)
- cloning and building of `yarp` and `icub-main`

pattacini, 2015-06-19:
For `travis` we had to rely just on `icub-common` for the dependencies plus we had to update `cmake`.

`travis` is now building smoothly!

closed 2015-06-19 by pattacini