robotology-legacy/wysiwyd #14: [iol2opc] Use segmentation algorithm based on stereo

Tobias-Fischer, 2015-04-27:
Hi all,

The current segmentation algorithm to extract blobs is based on luma-chroma segmentation. As we have found in the integration meeting in Sheffield, this method does not work particularly well. This is a shame, as the object recognition itself works very nicely.

One alternative with more promising results would incorporate the 3D information we get from the SFM module. The Point Cloud Library offers several algorithms to achieve what we want. The most promising seems this:
http://pointclouds.org/news/2012/04/03/new-object-segmentation-algorithms/
Some other algorithms also implemented in PCL are described here:
http://robotica.unileon.es/mediawiki/index.php/PCL/OpenNI_tutorial_3:_Cloud_processing_%28advanced%29#Segmentation
http://pointclouds.org/documentation/tutorials/region_growing_segmentation.php#region-growing-segmentation
http://pointclouds.org/documentation/tutorials/conditional_euclidean_clustering.php

As we rely on a quite noisy source, we might want to use outlier removal prior to the clustering:
http://pointclouds.org/documentation/tutorials/remove_outliers.php#remove-outliers
http://pointclouds.org/documentation/tutorials/statistical_outlier.php#statistical-outlier-removal

If someone else knows other libraries which can achieve similar results, please add them below.

Ugo @pattacini, what do you think?

Best, Tobi

pattacini, 2015-04-27:
I definitely agree to rely on stereo information for that.
Some stray thoughts:

- PCL works great - as per the video above -, however it does represents an heavy dependency (not that easy to provide/compile - I might be wrong any how, but I saw people struggling in the recent past). Could we achieve a segmentation nicer than what we currently have without PCL, for instance by handling the disparity information? Then, PCL can be added as a plus. Is it so clear that I don't like the experience I felt with PCL around :smirk: ?
- I'd suggest to develop code in [robotology/segmentation](https://github.com/robotology/segmentation) as e.g. a library and then instantiate  proper modules to talk to [iol](https://github.com/robotology/iol) part.

/cc @vtikha @alecive

stephane-lallee, 2015-04-27:
If I can suggest you to look into the "superpixel" algorithm developped at MIT. It should be part of the latest opencv releases and from what I've seen it works quite well.(it uses 2D though, not point cloud...)

-----Message d'origine-----

pattacini, 2015-04-27:
thanks @stephane-lallee, @vtikha has been already looking at **superpixel**; it's 2d and thus can be faster than any processing of point clouds. The only caveat, at present, is that `OpenCV 3.0` brings about compilation problems into our repos.

Tobias-Fischer, 2015-06-25:
Hi @pattacini and @vtikha,
do you have any news on using superpixel for the segmentation? As far as I know, OpenCV 3.0 now compiles fine thanks to https://github.com/robotology/yarp/commit/5665c99135fa0ba26480a617c91bbfb90f82bb87.

Thanks, Tobias

pattacini, 2015-06-25:
Hi @Tobias-Fischer 
We've just started playing around with the new stereo-disparity, and @vtikha has been a bit busy with other projects; however, he's got already something new that hopefully will improve the segmentation. If to use the stereo is still to be decided though.

Perhaps, @vtikha can add more info on that.

vtikha, 2015-10-28:
Hi @Tobias-Fischer, I have tried a few ways of going about this and just committed a new module to the IOL repository that greatly enhances the detection and segmentation of objects. For now this is still in 2D but I have also started working on a 3D version. The module is called `lbpExtract` and uses Local Binary Pattern to analyse the scene and extract blobs which are then finely segmented. In a nutshell, It is a powerful discriminator and does not cost a lot in terms of computation and it is less sensitive to changes in illumination than many other descriptors. You can find a better description and all the documentation [here](http://robotology.github.io/iol/doxygen/doc/html/group__lbpExtract.html).
This has been fully tested on the robot with IOL and we are very satisfied with the results. 
Here is a quick video that we made yesterday of it simply running and and image of the module running within the IOL demo. 

[image]

[image]

Tobias-Fischer, 2015-10-29:
Hi @vtikha,

we gave `lbpExtract` a quick try, and it seems to work very nicely! Thanks for this addition, I am sure this will be helpful for future research :).

@pattacini @gregoire-pointeau @maxime-petit @jypuigbo @uriel-martinez: I've created a new XML script for `iol2opc` employing `lbpExtract`, please give it a try as I suppose we should start using this system from now on: https://github.com/robotology/wysiwyd/commit/124860cc8f5645275f0720cb22a1507918591a59

Best and thanks again,
Tobi

vtikha, 2015-10-29:
Hi @Tobias-Fischer Great! You are welcome, I am happy you are finding it useful. Let me know if you have any issues, questions, enhancements etc... 
Ciao!

Tobias-Fischer, 2015-10-30:
Hi @pattacini, should we close this issue?

pattacini, 2015-10-30:
yes indeed, I think so :+1:

closed 2015-10-30 by Tobias-Fischer