robotology-legacy/wysiwyd #16: [iol2opc] set m_present=false when object gets out of sight

Tobias-Fischer, 2015-04-30:
Hi,
there is an issue with iol2opc. Thus far, adding objects in the OPC works nicely. However, if an object gets out of the scene, `m_present` is still set to `true`. We therefore need to find a way to
a) detect once an object is removed from the scene
b) find this object in the OPC and set `m_present=false`

The issue is that we don't keep track which objects are added to the OPC. I think we need to create a list with known objects, and keep this list up to date. What do you think, @pattacini?

Best,
Tobias

Tobias-Fischer, 2015-04-30:
Related to this is another issue. I was about to create a method to remove all objects. The classifier provides this ability by sending the command `forget all`. This is what I have done in https://github.com/robotology/wysiwyd/commit/274bdfd3d9b466cc0cfd82695b081508d7ce5e3f

However, that does not remove the objects from the OPC either.

I just found that https://github.com/robotology/himrep/blob/master/modules/linearClassifierModule/src/linearClassifierModule.cpp offers the command `objList`. If that was transparent through `https://github.com/robotology/iol/blob/master/src/himrepClassifier/classifier.cpp`, one could access the list of objects in iol2opc. Then, at least all objects could be removed from the OPC.

Tobias-Fischer, 2015-04-30:
And another issue: For now, several (physical) objects in the scene can be tagged with the same name. However, there is only one object in the OPC per name. Therefore, the position etc. of the OPC object is "random" (i.e. depends on the position in the `blobs` list.
Therefore, I think it would be better if we only associate each object in the object database to exactly one object in the scene (i.e. the one with the highest score).
If I remember correctly, this was originally the case in iol, right? @pattacini if you could please point me to the corresponding code, and I can go ahead and implement this in iol2opc if you want.

Thanks,
Tobias

pattacini, 2015-04-30:
The way I'd go for is the following:

1. Upon connection between 'iol2opc' and 'himrepClassifier', we ask for the list of currently known classes by means of the "`list`" rpc command (I've just added it up). This way, we can build the internal database we have to maintain later on.
We can attach a callback to the connection event by relying on the `PortMonitor` class. Be careful: don't do any access to the port within the callback, but just schedule to do the processing on a different thread (e.g. in the `updateModule()`).

2. At run-time we can recognize objects available in the scene and thus tick them off the list we have built. All the remaining objects need then to be handled such that their `m_present` attribute is disabled in `OPC`. To this end, I'd filter out some fluctuations by disabling only objects that are persistently unavailable for more than 1 second.

For the last issue, I'd point you at how the original [`findName()`](https://github.com/robotology/iol/blob/master/src/iolStateMachineHandler/src/classifierHandling.cpp#L310) function was coded.

PR #17 (2015-05-05): fix for #16
This should do pretty much what is required to fix #16.
@Tobias-Fischer, give it a try if you'll be willing to.
closed 2015-05-05 by pattacini