robotology-legacy/wysiwyd #18: [ARE] use depth2kin

Tobias-Fischer, 2015-05-07:
Hi all,

as we all know, we have started using stereo vision to get the position of objects. So far, we directly employ the 3D point of the object acquired from SFM to push/point/grasp the object. However, as far as I understood @pattacini correctly, we should "correct" the 3D point using depth2kin.
The actual implementation is straightforward (see https://github.com/robotology/grasp/blob/master/precision-grasp/graspSynthesis/src/precisionGrasp.cpp#L1133). However, I am not sure yet where the correction should be made. Directly in subsystem_ARE? But what if someone still uses the reactable - then the 3D coordinates should be used without correction. Shall we introduce a boolean flag to specify whether the point should be corrected or not? If so, should there be a default (which one)?

I think this is especially important for the stuff you are doing at Inserm, @AnneLaureM, @gregoire-pointeau.

Best, Tobias

gregoire-pointeau, 2015-05-07:
When we call ARE, we do it with the position given by the OPC. If the stereo vision has to correct a position, it should be done in the OPC.
For now, the subsystem ARE takes X Y Z position (that we extrat before from the OPC).

I talk with Ugo, to modify the subsystem to use either the X Y Z, or directly the OPC object.

pattacini, 2015-05-07:
@gregoire-pointeau there is a caveat in that corrections are arm specific (and thus it's very likely they will be different), therefore `OPC` should contain only the 3D points provided by `SFM`, a sort of common substrate. As soon as we're asked to perform a motor action (i.e. in `ARE`), we take those points to then apply the correction on top.

Tobias-Fischer, 2015-05-07:
@gregoire-pointeau: Sorry, I forgot to mention that we can't put the points directly in the OPC because of the reasons @pattacini mentioned.
The easiest solution I can think of is extending `take`, `push`, `point` and  `dropOn` with a `bool correctPosition`. If `correctPosition==true`, we apply the corrections by depth2kin. Otherwise, we directly use the supplied `target`.
There are two decisions in that case: should `correctPosition` have a default value? What should happen in case `depth2kin` is not available?

Best, Tobias

Tobias-Fischer, 2015-05-07:
Hi, 
after talking to Maxime we came up with another option. We could overload the methods in `ARE` to expect an `Object`. Then, if the object can be casted to an `RTObject`, we could apply some offsets similar to the ones in `reservoirHandler`. If it can be casted to a to be defined `stereoObject`, we could apply the `depth2kin` corrections.
In fact, icubClient already offers methods which can be called with the name of an object, and then calls `ARE` to perform the actions. This could also be a good point to check the `Object` type.

Best, Tobias

pattacini, 2015-05-07:
But then the code will undergo somehow a _biforcation_: we will have snippets relying on `RTObject` and snippets relying on `Object`, while we should abstract from this detail. Unless the objects will be created dynamically, making use of factory methods that in turn will get called depending on the actual value of a configuration option (stored in a file). But if the latter will be the case, we could read this option within  the `subSystem_ARE` ... and that's all folks! This way, we could also avoid specifying that further parameter in the _push/take_ actions' signatures.

pattacini, 2015-05-08:
To expand on my previous post, will we be finding ourselves in scenarios where the robot has to deal with `RTObject` and `stereoObject` at the same time?

- **`yes`**: the solution proposed by @Tobias-Fischer and @maxime-petit is more thorough and we should pursue this way.

- **`no`**: configuring `subSytem_ARE` at start-up from a option stored in a file would be easier and as effective as the previous solution.

Answering the question "_what shall we do when `depth2kin` is not available?_": we should use the 3D point provided by `SFM` - as it is. Any contraindication?

Tobias-Fischer, 2015-05-08:
Hmmm .. maybe we can discuss this stuff in our next integration meeting, to see what is needed and how we should pursue. With this issue I mainly wanted to make people aware that the corrections by `depth2kin` for now need to be manually applied.
Unfortunately, you lost me with the factory methods etc. - I don't see why they would be needed. In our proposed solution, as a user, you would not need to do anything. All you do is giving a name of an object you want to point to etc., and iCubClient figures out which object type it is. As the objects are only created by `reactable2opc` and `iol2opc`, these are the only locations where the proper object type would need to be created. Maybe I am missing something though ;).

Best, Tobias

pattacini, 2015-05-08:
Ok I see, you're right, I was missing the part regarding the place in the architecture where objects are created. If everything is under control and limited to `rt2opc` and `iol2opc` we're done then.

Tobias-Fischer, 2015-05-08:
Okay cool. @gregoire-pointeau: Are you happy to directly call the iCubClient with the name of the object, rather than ARE with a vector of the position of the object? Do you know who else is using this stuff? And I saw that you are doing some corrections for the reactable objects - could you please let me know which ones?

I am happy to implement this in the next week or so, as it should be straightforward.

Best, Tobias

Tobias-Fischer, 2015-08-29:
Hi @gregoire-pointeau and @pattacini,
Maybe you guys can have a look at this at BCBT?

Best, Tobi

pattacini, 2015-08-30:
@Tobias-Fischer I'm not attending BCBT...

Tobias-Fischer, 2015-08-30:
Ahhh yeah I remember you said you won't go to BCBT. I guess it's not too urgent anyways, maybe I'll have a look into this issue as soon as I have some spare time ;).

Tobias-Fischer, 2015-12-02:
@pattacini: Just a reminder - maybe we can have a look tomorrow.

pattacini, 2015-12-02:
:+1: yep

pattacini, 2015-12-12:
I think we could solve this issue more elegantly as follows:

1. `depth2kin` is running, loaded with the experts and connected to our system. Then, we can easily ask for the mapping. We are thus using the **iCub vision**.

2. Two possible cases are given for the configuration with the **Reactable**:
    2a. `depth2kin` is running with no experts and connected to our system. Then, any query will return exaclty the same input point.
    2b. `depth2kin` is not running. Then, we can detect this condition and skip the query.

This way we don't have to create _ad-hoc_ new objects.

I've also realized that the communication with `depth2kin` should somehow take place at the `ARE` level, since it may happen that `SubSystem_ARE` is called without specifying the hand, leaving this task to `ARE` itself.

@Tobias-Fischer @maxime-petit, how do you see this solution?

pattacini, 2015-12-12:
Further, if the change will be located in `ARE`, this means that we don't have to do anything here.

Tobias-Fischer, 2015-12-13:
Hi @pattacini,
what would happen if someone provides the already corrected coordinates to `ARE`? These would be "double-corrected" leading to wrong coordinates. It's kind of breaking the compatibility if this change is made directly in `ARE` for other people, unless there is a flag which is set in a way that by default already corrected coordinates are assumed.
Other than this your suggestion sounds great to me :).

Best, Tobi

pattacini, 2015-12-13:
I was thinking about introducing that right flag, actually :smirk:, to address the condition you mentioned.

Anyway, `ARE` has already the capability to apply some internal correction (e.g. via _kinesthetic teaching_), thus this change won't totally break the original behavior, up to some extent.

Tobias-Fischer, 2015-12-13:
Sounds good to me then! Thanks :)

Tobias-Fischer, 2016-01-21:
Hi @pattacini,
just checking whether this is still on your radar ;).

Best, Tobi

pattacini, 2016-01-21:
Yeah :smirk: 
However I've changed my mind again on the way to go ...
... I would finally choose a solution embedded in `subSystem_ARE`.

The rationale is that in the end I don't want to overcrowd `ARE` too much.

Tobias-Fischer, 2016-01-21:
Sounds good :)

pattacini, 2016-03-07:
Lately, we have coded an [**alternative**](https://github.com/robotology/iol/tree/master/src/iolReachingCalibration) to `depth2kin` which has been specifically designed for calibrating reaching in a table-top scenario (typical _IOL_ demo).

It could be worth investigating in Lyon if this solution turns out to be simpler.

pattacini, 2016-03-17:
In the end we went for #83

closed 2016-03-17 by pattacini