robotology-legacy/wysiwyd #19: [wrdac]: no headers tree

pattacini, 2015-06-18:
It seems that presently it is not possible to export multiple levels trees of libraries headers. This applies specifically to `wrdac` where we have include branches like `clients`, `knowledge` and `subsystems`. Building is not problematic, whereas installing triggers this issue.

Temporary solution: we don't install `wysiwyd`
Proper solution: we refactor the huge `wrdac` (it's too big indeed) adhering to the above standard.

/cc @Tobias-Fischer

stephane-lallee, 2015-06-18:
Hi,

Just a word: it should be pretty straightforward to "cut" wrdac in 2/3 different libraries... The dependencies are: 
- clients & subsystems are interdependent, 
- they both depend on knowledge.

One thing I wanted to do at some point was to turn all the subsystems into proper thrift interfaces with the modules in question. It should not be hard, but it may be a long and tedious task...

-----Message d'origine-----

pattacini, 2015-06-19:
Hi @stephane-lallee! How are you doing?
Great then, we can really think to refactor `wrdac` in the way you've suggested.

Tobias-Fischer, 2015-06-19:
Hey @stephane-lallee and Ugo @pattacini,
the reason why I used multiple exports is that I want to use WYSIWYD as library in another project. Originally, all include files got exported to the same folder (or some not at all?!?), which caused trouble compiling the other project. On Linux, the multiple exports work nicely (another reason to deprecate Windows :P).
Splittling wrdac in multiple libraries sounds like a great idea, but who got the time to do it?

Best,
Tobias

pattacini, 2015-06-19:
@Tobias-Fischer working fine doesn't exclude it is an hack, which seems to be doable only in linux.
Unfortunately we have to provide support for windows too.

I'll do the job (but without `thrift` :smirk:).

pattacini, 2015-06-19:
I opted for the quickest solution since it allowed me to use/debug the pre-existing options of the macro.

commit da0e7f61 (2015-06-19): [wrdac]: in the end we can employ the FILES_WITH_PATH option of the macro (update of icub-main is required); fix #19
closed 2015-06-19 by pattacini