robotology-legacy/wysiwyd #22: AppVeyor fails compiling due to cmake error

pattacini, 2015-08-30:
We're experiencing the following cmake error while checking `wysiwyd` compilation by means of `AppVeyor` CI tool (thanks @Tobias-Fischer for spotting). This error showed up only recently, while compilation was used to go fine with the same binaries in the past.

```
Make Error at c:/Program Files (x86)/robotology/icub-1.1.16/cmake/icub-config.cmake:29 (include):
  include could not find load file:
 
    __NSIS_ICUB_INSTALLED_LOCATION__/lib/ICUB/icub-export-install-includes.cmake
Call Stack (most recent call first):
  CMakeLists.txt:13 (find_package)
```

@mbrunettini @drdanz 
- Windows binaries have been updated somehow?
- Do you have any clue?

pattacini, 2015-08-30:
Compiling `wysiwyd` from `yarp` and `icub-main` sources is smooth.

Tobias-Fischer, 2015-08-30:
Just one small correction: The Windows binaries of `yarp` and `icub-main` have changed since the last successful compilation. We used `yarp_2.3.64_v12_x86_0.exe` previously (which is not available anymore), and changed to `yarp_2.3.64_v12_x86_1.exe`. Same goes for `icub-main`: From `iCub_1.1.16_v12_x86_0.exe` to `iCub_1.1.16_v12_x86_1exe`.

pattacini, 2015-08-30:
Ah ok, then this is another hint suggesting there something strange with the latest binaries...

Tobias-Fischer, 2015-08-30:
@mbrunettini could it be related to https://github.com/robotology/icub-packaging/commit/4c050e0ad0c02c70fedfb38f3b8f88cae563e5ea ?

mbrunettini, 2015-09-15:
The binaries have been rebuilt recently to add missig cmake files.
It seems that a variable (__NSIS_ICUB_INSTALLED_LOCATION__) is not resolved correclty.
I'm going to investigate on this issue and let you know.

mbrunettini, 2015-09-15:
The problem is with NSIS and AppVeyor: the icub and yarp packages are installed with the "/S" parameter, which runs sinlently the installer IN BACKGROUND. This causes the command to return before the installation is completed (strange, but you can try). So when the compilation is executed the icub packaged installation is not yet completed, in fact the  __NSIS_ICUB_INSTALLED_LOCATION__ is resolved at the end of the installation process.
Is it possible to pause the Appveyor script for a while?

drdanz, 2015-09-16:
Perhaps it is something that should be fixed in the package?
See http://nsis.sourceforge.net/Docs/Chapter4.html#silent

mbrunettini, 2015-09-16:
@drdanz : I can't see, in the document you posted, anything that prevents the silent installer to run in background..

mbrunettini, 2015-09-16:
I found a fix: the AppVeyor script must be changed as follows
The following two lines (line 8 and 9)
    c:\dev\yarp.exe /S
    c:\dev\icub.exe /S
must be replaced bye the following
    start "" /WAIT "c:\dev\yarp.exe" /S
    start "" /WAIT "c:\dev\icub.exe" /S

Tobias-Fischer, 2015-09-16:
Hi @mbrunettini,
Thanks for your help! I hope we can get this issue resolved.
Unfortunately, your trick did not work  :( (https://ci.appveyor.com/project/pattacini/wysiwyd/build/1.0.947). Do you have any other ideas?

Best, Tobias

mbrunettini, 2015-09-16:
Try using a longer timeout, no other ideas..

Matteo

Tobias-Fischer, 2015-09-16:
Hmm, I tried a three minutes delay, and it still doesn't work: https://ci.appveyor.com/project/pattacini/wysiwyd/build/1.0.949

Additionally to the three minutes, several other modules are compiled, so there is at least a five minutes difference between installing icub-main and the execution of the cmake which fails.

It's also weird that it never happened with the old binaries, and there we did't have a delay at all.

mbrunettini, 2015-09-16:
Now the installer needs to replace a string in a file with a path that is available only at the end of the installing process (it depends on the installation path).
Normally, it’s replaced correctly.
Inside the AppVeyor environment is untouched.
Matteo

pattacini, 2015-09-17:
Can it be that it's right this last operation done by the installer that is failing in the `AppVeyor` environment? This would explain why the timeout doesn't the job.

@mbrunettini could you provide info about the file and the string under exam so that we could insert checks in the `AppVeyor` script?

mbrunettini, 2015-09-17:
The file is
c:\Program Files (x86)\robotology\icub-1.1.16\cmake\icub-config.cmake

In this file, the following label
``__NSIS_ICUB_INSTALLED_LOCATION__``
must be replaced by
c:\Program Files (x86)\robotology\icub-1.1.16

pattacini, 2015-09-17:
thanks, we'll investigate it

Tobias-Fischer, 2015-09-17:
Strange things are happening: https://ci.appveyor.com/project/pattacini/wysiwyd/build/1.0.957

The string ` __NSIS_ICUB_INSTALLED_LOCATION__` does not appear in `c:\Program Files (x86)\robotology\icub-1.1.16\cmake\icub-config.cmake`, but `cmake` somehow thinks it does ...

Tobias-Fischer, 2015-09-22:
Hi @pattacini, @mbrunettini,

I found the devil and reported the bug in https://github.com/robotology/icub-packaging/issues/1. It would be great if this could be fixed soon, as it affects all Windows systems. I am surprised no one else reported this bug yet.

Best, Tobias

mbrunettini, 2015-09-28:
Bug fixed and new packages updated.

Tobias-Fischer, 2015-09-28:
Thanks @mbrunettini! Working fine now.

closed 2015-09-28 by Tobias-Fischer