robotology-legacy/wysiwyd #23: Change to travis container builds

Tobias-Fischer, 2015-09-08:
Hi,
recently we started getting warnings about changing to travis container builds rather than using their legacy infrastructure. Apparently, it's faster, allows caching, ... (see http://docs.travis-ci.com/user/migrating-from-legacy/?utm_source=legacy-notice&utm_medium=banner&utm_campaign=legacy-upgrade).
The only drawback is that `sudo` cannot be used. We use `sudo` to
1) install the `icub-common` package
This will require adding the `deb http://www.icub.org/ubuntu precise contrib/science` in their new whitelist (https://github.com/travis-ci/apt-source-whitelist), which in turn needs a GPG key, which as far as I know does not exist yet.
2) install `yarp`, `icub-common`, `icub-main` and `kinect-wrapper`. To avoid using `sudo` there, we can set the corresponding environment variables and install the packages there rather than in `/usr/bin` or wherever they get installed by default

I don't think it's urgent to switch, but probably it needs to be done sooner or later. Ugo @pattacini, do you know who is in charge of the `deb` packages and could create a GPG key? I guess that's a long missing feature anyways ..

Best,
Tobi

pattacini, 2015-09-09:
Probably @mbrunettini could help in this respect.
/cc @drdanz @lornat75

drdanz, 2015-09-09:
+1 for adding a signing key to the repository... We have already been discussing this with @mbrunettini and it should not be a very complicated thing to do, but unfortunately time is never enough!

lornat75, 2015-09-10:
fine by me too.

Tobias-Fischer, 2015-09-28:
Now that https://github.com/robotology/wysiwyd/issues/22 is closed, are there any news on the signing key? :)

Tobias-Fischer, 2015-11-02:
Hi @drdanz and @mbrunettini,
have you further discussed whether it would be possible to add a signing key for the yarp/icub repository?

Best, Tobi
/cc @pattacini @lornat75

traversaro, 2015-11-19:
FYI : https://github.com/travis-ci/docs-travis-ci-com/pull/390

mbrunettini, 2015-11-20:
To add the key signature to an already populated repository is not possible, so this is not an easy task.
But why do you need the signature?
You can easily skip the "are you sure?" question by adding an
--yes  --force-yes
in the apt command. Es:
 apt-get --quiet --yes update
 apt-get --quiet --yes --force-yes install icub-common
(the "--quited" is not mandatory, but it's useful in scripts..)

Tobias-Fischer, 2015-11-20:
Hi @mbrunettini,
the original reason why I opened this issue was that to use the new container builds of travis, you cannot use `sudo` anymore (and thus, not `apt-get`). To install packages, you rather have to add them to a whitelist (for which a signed key is necessary), and you can then install them using some travis specific commands.

As @traversaro pointed out, it seems like the request to add more sources to this whitelist is currently stopped. I am not sure for what reason and when it will resume.

However, I think there are some good other reasons. First, security. How can I be sure that the `icub-common` package which I install with `--force-yes` is actually created by you? What if someone gains access to your server and puts another harmful package on there, which in worst case deletes a lot of research data and could also harm some iCubs. I know, it's not likely, but crazy things happen on the world nowadays.
Second, Ubuntu does not update packages with the graphic interface in case some of them have to be approved because they are not signed. For command line users, this is not an issue (besides the security question asked), but for people relying on the graphical tool it can be a serious issue.

If it's hard to add a key to an existing repository, maybe a new repository could be created to support upcoming versions of Ubuntu / Debian?

Best, Tobias

mbrunettini, 2015-11-20:
I think the security issue is not a big issue.. 
To add another repository for new packages means make it available to another address (or to change  the address of the old one), so losing back compatibility and asking to all people to change their packages list files, this could be a worse problem.

Tobias-Fischer, 2015-11-20:
People will have to change their package list files anyway when they move to a new distribution .. So when Ubuntu 16.04 will be released, the new (signed) repo could contain files for this distribution, while the old (unsigned) repo continues to provide updates for older supported distributions.

Best, Tobias
/cc @pattacini @lornat75 @drdanz

drdanz, 2015-11-20:
@mbrunettini I agree with @Tobias-Fischer, unsigned packages are a **big** security issue and we should fix that as soon as possible. Moreover it is also 1) annoying and 2) a very bad presentation to people approaching our software. Personally, as a debian user, I already hate installing software that is not on debian repositories. if I wanted to try YARP from binaries, I would immediately give up after seeing the unsigned package.

Making signed packages with reprepro should be just a matter of creating a gpg key and [adding a line to reprepro configuration](https://wiki.debian.org/SettingUpSignedAptRepositoryWithReprepro). This won't change the packages that are already uploaded but will automatically sign the new packages.

Tobias-Fischer, 2015-12-11:
Hi @mbrunettini,
Pushing this issue as I fully agree with @drdanz. Especially as it only seems to be a small change without any impact on the existing packages, but big improvements for future packages.

Best,
Tobias
/cc @pattacini @lornat75 @drdanz

lornat75, 2015-12-22:
I spoke with @mbrunettini: one problem is that it seems that the repository we set up can't provide mixed signed/non-signed packages, therefore signing packages may require reinstalling the repository or signing old packages. Anyway @mbrunettini will try to find a way to sign new packages. This is scheduled for the first months of next year.

pattacini, 2016-02-29:
Issue moved to [robotology/icub-packaging #3](https://github.com/robotology/icub-packaging/issues/3) via [**ZenHub.io**](https://www.zenhub.io/)

closed 2016-02-29 by pattacini