robotology-legacy/wysiwyd #24: Calibration of Kinect without employing Reactable

Tobias-Fischer, 2015-10-06:
Hi all,

thus far we have employed the Reactable to get the transformation matrix from the robot root reference system to the Kinect reference system. It would be great if we replace this by a solution which employs stereo vision.

I see two possible solutions:
1) Use object detection employing both, the Kinect camera as well as the iCub eyes. If we a sufficient amount of objects not being planar etc. etc., we should be able to compute a transformation matrix.
2) Point cloud alignment: Align the point clouds acquired by the Kinect with that of the iCub eyes. We could use ICP for that, and employ prior knowledge (an approximate transformation is known from the geometrical layout of the Kinect pole).

Best, Tobi

pattacini, 2015-10-07:
I believe that a first and quite effective implementation for that task would be to click on a `yarpview` to get then a 2d point for querying the corresponding 3d point through `stereo-vision` and then click the same 2d point on the Kinect and retrieve a second 3d point. We will finally build two sets of 3d points we can use to run the software we already have that finds out the desired matrix transformation.

This approach will put the object detection aside, which needs to be somehow developed for the Kinect device, I think.

Tobias-Fischer, 2015-10-07:
Sounds great!

maxime-petit, 2015-10-07:
We may also reuse the code for the kinect/reactable calibration when we needed to click on an object in a `yarpview` (or a direct window?) showing the depth of the kinect. The code is in `agentDetector` (line 300 in the cpp file).

After the click event, the x/y/z of the only object on the reactable was extracted : we can have the only object recognize by the stereo-vision instead?

Tobias-Fischer, 2015-10-07:
Good call @maxime-petit! That might be even easier. All we would need to do is changing `RTObject` to `Object`:)

stephane-lallee, 2015-10-07:
Hi guys,
If you dig a little bit the method proposed by Ugo was already implemented in EFAA (reactableVisualCalibrator) or as Maxime said it should be pretty straightforward to extract the relevant code from agentDetector.
In theory referenceFrameManager should be very easy to implement any new way of calibrating a device against another.

 To bad we never managed to publish anything about it... If somebody wants I have a draft ready to be revised (I gladly give away the first authorship 😉)

Cheers,
Stephane

-----Message d'origine-----

pattacini, 2015-10-07:
@stephane-lallee I remember that paper! :smile:

pattacini, 2015-10-07:
The title was great:

"_**One to bind them all**: n-sensors calibration framework for intelligent spaces and robotic setups_"

Tobias-Fischer, 2015-10-07:
Just out of interest: is it similar to http://wiki.ros.org/Papers/TePRA2013_Foote?action=AttachFile&do=view&target=TePRA2013_Foote.pdf ?

pattacini, 2015-10-07:
@Tobias-Fischer from the abstract it seems that the **_tf_** library implements only a sort of database of coordinates systems, whereas our approach was also aimed to find out the best homogeneous transformation (possibly along with anisotropic dilation/contraction) that relates two point clouds.

The code is documented here: http://wiki.icub.org/iCub/main/dox/html/group__MatrixTransformations.html

stephane-lallee, 2015-10-08:
Yep, the title was the best part in this paper. ^^'
Should we try to submit this again somewhere ? Maybe it could be extended
with your current experiments... Or maybe is it better to not wake up what
as been sleeping in darkness for so long...

Stéphane Lallée

*-----------------------------------------------------*
A*Star Scientist
Institute for Infocomm Research - I²R <http://www.i2r.a-star.edu.sg/>

*Fusionopolis, Tower South 10#104**Singapore*
*(+65) 6408 2074*
stephane-l@i2r.a-star.edug.sg
*------------------------------------------------------*

maxime-petit, 2015-10-08:
Yop,

I guess it also depend on how advanced the paper was. I will try to see `kinect`/`stereo` in the next days. May have to play a bit to remember exactly how agentDetector calibration was used :dancer: 
From memory it is automatic as soon as the [kinect] matrix is not present in `RFH`, isn't it?

./Max

stephane-lallee, 2015-10-08:
Yes... If the kinect was not available in RFH then agentDetector would be waiting for input from yarpview. I think you have to run yarpview with an optional parameter to allow click forwarding though, it should be in the xml.

 I attached the draft of the pdf in the last mail in case you want to take a look.

-----Message d'origine-----

maxime-petit, 2015-10-08:
Hum... Same player try again?

Tobias-Fischer, 2015-10-08:
Fixed with https://github.com/robotology/wysiwyd/commit/154b648d2192e4a40c84190daf88698e6bed0fe1 and https://github.com/robotology/wysiwyd/commit/dd468d8815d1b67c56dd6c62c3f9d8d7cc65477d

Thanks @maxime-petit!

closed 2015-10-08 by Tobias-Fischer