robotology-legacy/wysiwyd #25: Speech recognition problems in proactiveTagging and ears

jypuigbo, 2015-10-07:
During BCBT and IROS we have detected in Barcelona an annoying bug/issue when using speech recognition. 

1) When we try to run the demo, before running any of the other modules, ```speechRecognizer``` already has a grammar loaded, which detect yes, no, let's play pong, etc. (Like the sentences used in EFAA?). This interferes with other modules that use speech recognition. Has anyone faced this same problem or is something particular here in Barcelona? Do you know if there is any method in speech recognition that automatically loads a grammar? If not, I will check which could be the interfering module. 

2) Both, ```ears``` and ```proactiveTagging``` are error-prone to the first issue, as they segFault when they receive an unexpected sentence from the ASR or they output invalid values, depending on the case. I guess that this issue is easy to deal with. I can take care of it if you agree that the changes should be done directly on these modules.  

Let me know what you think of it. 

Jordi

Tobias-Fischer, 2015-10-07:
Hi Jordi,

regarding 1) as far as I know, we have the same "issue". But @maxime-petit will know for sure.
For 2) What is the ASR? And yes, `ears` and `proactiveTagging` should be changed so they can handle unexpected sentences.

Best,
Tobi

maxime-petit, 2015-10-12:
Yop!

Sorry I forgot this email :dancer: 

1) The default grammar is loaded as startup by `speechRecognition` but is actually bypassed as soon as you provide your own grammar. So it should not cause any trouble. The only known issue is that actually when you provide a new grammar, the system await a possible "sentence" to be finished before changing the grammar. So you should check that your microphone is mute (or at least reduce the sound amplifier), otherwise the speech recognizer might heard background noise, think that you are speaking and then not loading the defined grammar properly.

Note that in my windows setup, `speechRecognizer` always increase at 100% the amplifier, so check this value also after launching it

You should check that you have a line saying "you can speak" or something similar, from memory with a big line full of "=======================================".

We might however try to remove this default grammar. I think the purpose was to have generic recognition sentence that may be recognize at anytime, like "icub stop!". If the mute microphone is not working/suitable, you might want to edit the default grammar directly. It should be in the configuration of speechRecognition, `smallGrammar.xml` by default.

2) I dont understand because it happened to me sometimes and I think I have put protection in proactiveTagging to handle unexpected sentence coming from the speech. Can you provide an exact example of the sentence and the exact part in `protactiveTagging`. Maybe not 100% of the recognition is protected and it should be dealt with indeed.

pattacini, 2015-10-12:
1. Would't it be easy to provide an empty grammar file with which running the module? By the way, we are using `speechRecognition` along with **iol** with no problem since ages and we do load new grammars at start-up; as far asa I know, we do not need to wait, or at least I've never taken this into consideration thus far.

maxime-petit, 2015-10-12:
1. Yes the empty grammar file was also my suggestion, I may not have been clear enough. I have written "remove" but I was thinking removing the text inside the grammar, not the grammar per se. But maybe it can cause trouble to the `speechRecognition` that might need at least one proper grammar. I ll check that.

And yes indeed it should be fine, at least if the microphone is setup properly (no over-amplification, not to noisy background, ...). In that case there is actually not "waiting" time. I only had this with like embedded laptop microphone with lot of people behind.

maxime-petit, 2015-10-12:
Yop,

The `speechRecognizer` require a valid grammar to run : it should be at least one RULE NAME part. You can try to remove everything and just let a very long and complicated sentence to not trigger it. I will try to look if it is possible to remove this requirement. 

But again, this should not be an issue because every grammar sent to speechrecognizer overwrite this one.

maxime-petit, 2015-10-16:
@jypuigbo : do you have news on this? is it better after some microphone tuning/default grammar editing?

Tobias-Fischer, 2015-10-30:
@jypuigbo: Any news?

closed 2015-11-10 by pattacini