robotology-legacy/wysiwyd #27: Conflic between ARE and attentionSelector

gregoire-pointeau, 2015-10-27:
Hello we have a problem that I guess come from conflicts between ARE and attentionSelector, when using the subsystem_ARE.

What we want to achieve is: the robot looks in front of him at first. There are 4 objects: top/bottom right/left. for each object, the robot looks the objects 2s, then point the object.
(The idea is then with reservoir computing to give the reservoir just the joint information of the robot corresponding to the look, and it should predict the joint sequence for the point.)
So we do the following:
    `bool bSuccess = iCub->look(sObject);`
    `Time::delay(1.0);`
    `bSuccess &= iCub->point(sObject, bHand, true);`

But the iCub does not look at first the object, but when he is supposed to point at it, it looks correctly.

Summary:when I ask only to look `iCub->look(object)`, the robot doesn't move. When I ask to point `iCub->point(object)`, the iCub looks at the object and point correctly.

maxime-petit, 2015-10-27:
Have you tried to run `attentionSelector` without ARE and check the look command?

Then in which order do you launch `ARE` and `attentionSelector`? `attentionSelector` first no? Can you try to switch the order?

I ll try to take a look during the afternoon. 

./Max

gregoire-pointeau, 2015-10-27:
attentionSelector without ARE works (well it works in the way that it will look at salient objects) but I didn't tried the explicit `iCub->look(object)` without ARE.

Concerning the order, I tried both, and none of them work.

stephane-lallee, 2015-10-27:
From what I remember there were some option for each ARE cmd that was basically telling to control or not the gaze.  
AttentionSelector and ARE were both interfacing directly to the gazeCtrller which may create conflicts indeed...
You should look into the "point" code to see what are the parameters.

-----Message d'origine-----

pattacini, 2015-11-02:
The only practical and neat solution is to make `attentionSelector` rely on `ARE` rather than issuing commands directly to the gaze controller.
I'll see to apply this change this week.

pattacini, 2015-11-09:
Addressed by #29

closed 2015-11-09 by pattacini