robotology-legacy/wysiwyd #28: [OPCClient] Invalid pointers

Tobias-Fischer, 2015-10-30:
Hi all,
as discussed on slack, we have issues with the `OPCClient`. See for example this code snippet which leads to an invalid pointer, as `checkout()` deletes all the `Entity*`, and allocates new memory for new `Entity*` objects:
```
Object* o = opc->addOrRetrieveEntity<Object>("test");
opc->checkout();
cout<<"This does not work, because checkout changes the pointers!!!" << o->name()
```

My comments on slack were as follows:
I think we have a fundamental issue. Every time EntitiesCache is used, a list of pointers to entities is returned. However these pointers might change over time (I am not sure why). So if a module uses EntitiesCache, and then iterates over the entities, the actual pointers to the entities might have already changed. And then we see corruptions, as we've seen yesterday in #bcbt-2015. So we need to figure out why the pointer to the entity with the same OPC id changes over time. You might think: Why not always use EntitiesCacheCopy? This is only feasible if a module is only reading attributes from the entities. As soon as an entity needs to be changed, you must use EntitiesCache. Otherwise you change the values of the copy, but not the entity itself.

In the meantime, as above, I have found that the `checkout()` is the evil method changing the pointers.

I guess the following should be discussed:
What should `checkout()` do by default? Should `updateCache` be really `true` by default? It deletes all pointers, and retrieves the entities fresh from the OPC. Isn't what we really want an update of the entity, i.e. the entity itself is kept (thus no deleting of any pointers etc.), but the attributes of the entity are updated with the content which is stored in the OPC?

Best,
Tobi

stephane-lallee, 2015-10-30:
Yes, checkout() was there only to get a fresh new snapshot of the opc. For updating existing entities you already have references to, update() was intended to be used.

Most of the time you don't need the whole opc content but just a few entities manipulated by your module, in this case you can just get those entities using addXXX or get and use update/commit.

Checkout was supposed to be used in very specific cases or when you don't know what you want to retrieve from the opc.

I remember the EntitiesCacheCopy was some method introduced by Grégoire to "break my castle" ;-)

-----Message d'origine-----

Tobias-Fischer, 2015-10-30:
Cool. Thanks @stephane-lallee! It seems like we found the evil method then! I propose we jointly go through all occurrences of `checkout()` and decide whether they should be replaced by `update()`. We should probably try doing this before the integration meeting.
The same goes for `EntitiesCacheCopy`, which in my eyes should only be used very rarely. Right now, I have the feeling it is used mostly because crashes due to invalid pointers can be avoided.

Best, Tobi

pattacini, 2015-12-12:
As side note, see also 506e2f5ac642ece0e8246ecf690846ced43e32ee.

The pointer returned by `OPCClient::getEntity()` was not checked against `NULL` causing `attentionSelector` to seg-fault whenever a call to `OPCClient::clear()` was done concurrently from within a different module (e.g. `test-cartesian-sim`).

This is to say that we should not take any assumption on the pointer returned by `OPCClient::getEntity()` and, further, it is clearly a good practice to always check against `NULL`, taking suitable actions in accordance to the result.

/cc @gregoire-pointeau @Tobias-Fischer @maxime-petit

Tobias-Fischer, 2015-12-13:
After your comment, I thought about this issue yet again. My previous suggestion of replacing `checkout()` with `update()` doesn't work, as we would never retrieve newly added objects from the OPC. However, replacing `checkout(updateCache=true)` by `checkout(updateCache=false)` might work. Actually, just replacing the default value of `updateCache` might do the trick. Or as said before, rewriting this damn module and reduce its complexity while doing some smart tricks would probably be the best (but time-consuming) solution ..

To get to your actual comment: Isn't it just delaying the race condition? What if `OPCClient::clear()` is called in between the check against `NULL` and the actual usage of the pointer?

Best,
Tobi

pattacini, 2015-12-13:
I'm honestly a bit lost among `checkout()`, `update()` and so on; I'd really need some time to catch up again with `OPCClient` ...

###### Further notes regarding your doubt on the race condition

Well, after the call to `getEntity()` you'll get a fresh local copy of the entity stored in the actual `OPC` only if the object is still contained therein (otherwise the pointer is `NULL`). Therefore, the returned pointer will always refer to a consistent local object, if the object did not disappear before the call to `getEntity()`.

Concluding, if a `clear()` is called in between the check against `NULL` and the actual usage of that pointer, there won't be any seg-fault but the robot will gaze at a location where the supposed target just vanished away.

Tobias-Fischer, 2015-12-13:
Regarding the doubts: I am not too sure what happens in the case to be honest. I think that a `clear()` between the check against `NULL` and the usage of the pointer would corrupt the local object, as `clear()` deletes the (object referenced by the) pointer. Thus, the memory which is referred by the pointer may still contain the object, but might as well be overridden by something completely different as the memory is freed. But I am not too deep into pointers etc. so that view might be wrong. That's why I try to avoid pointers unless I really, really think I need them.

pattacini, 2015-12-13:
I was referring to the case where `getEntity()` is called by module **A**, whereas `clear()` is called by module **B**. No problem here, since `OPC` will take care of performing concurrent actions in a thread-safe manner, let's say.

Instead, in the scenario where both `getEntity()` and `clear()` are called within the same module **A**, then it's the responsibility of **A** to protect against race conditions using proper mutex. It's not a fault of `OPCClient` here.

Tobias-Fischer, 2015-12-13:
Indeed, agreed :)!

Tobias-Fischer, 2016-03-16:
I haven't seen any issues regarding this for a while. At least the core modules seem to all stick to the "safe" policy. Well done everyone.

Closing here, but please don't forget about this in upcoming code.

Best, Tobi
/cc @robotology/wysiwyd-developers

closed 2016-03-16 by Tobias-Fischer