robotology-legacy/wysiwyd #30: verbRec warnings

pattacini, 2015-11-09:
Hi @MagnusJohnsson 
Thanks for committing the code the repo.

Could you please address the following warnings we have during compilation?

```sh
C:\dev\wysiwyd\main\src\modules\attentionRelated\verbRec\src\verbRec.cpp(214): warning C4244: '=' : conversion from 'int' to 'float', possible loss of data
C:\dev\wysiwyd\main\src\modules\attentionRelated\verbRec\src\verbRec.cpp(261): warning C4244: '=' : conversion from 'int' to 'float', possible loss of data
C:\dev\wysiwyd\main\src\modules\attentionRelated\verbRec\src\verbRec.cpp(262): warning C4244: '=' : conversion from 'int' to 'float', possible loss of data
C:\dev\wysiwyd\main\src\modules\attentionRelated\verbRec\src\verbRec.cpp(377): warning C4305: 'initializing' : truncation from 'double' to 'float'
C:\dev\wysiwyd\main\src\modules\attentionRelated\verbRec\src\verbRec.cpp(443): warning C4305: 'initializing' : truncation from 'double' to 'float'
C:\dev\wysiwyd\main\src\modules\attentionRelated\verbRec\src\verbRec.cpp(525): warning C4305: 'initializing' : truncation from 'double' to 'float'
C:\dev\wysiwyd\main\src\modules\attentionRelated\verbRec\src\verbRec.cpp(607): warning C4305: 'initializing' : truncation from 'double' to 'float'
C:\dev\wysiwyd\main\src\modules\attentionRelated\verbRec\src\verbRec.cpp(688): warning C4305: 'initializing' : truncation from 'double' to 'float'
C:\dev\wysiwyd\main\src\modules\attentionRelated\verbRec\src\verbRec.cpp(808): warning C4244: '+=' : conversion from 'float' to 'int', possible loss of data
C:\dev\wysiwyd\main\src\modules\attentionRelated\verbRec\src\verbRec.cpp(809): warning C4244: '+=' : conversion from 'float' to 'int', possible loss of data
C:\dev\wysiwyd\main\src\modules\attentionRelated\verbRec\src\verbRec.cpp(810): warning C4244: '+=' : conversion from 'float' to 'int', possible loss of data
C:\dev\wysiwyd\main\src\modules\attentionRelated\verbRec\src\verbRec.cpp(811): warning C4244: '+=' : conversion from 'float' to 'int', possible loss of data
C:\dev\wysiwyd\main\src\modules\attentionRelated\verbRec\src\verbRec.cpp(864): warning C4305: 'initializing' : truncation from 'double' to 'float'
C:\dev\wysiwyd\main\src\modules\attentionRelated\verbRec\src\verbRec.cpp(892): warning C4244: '+=' : conversion from 'float' to 'int', possible loss of data
C:\dev\wysiwyd\main\src\modules\attentionRelated\verbRec\src\verbRec.cpp(893): warning C4244: '+=' : conversion from 'float' to 'int', possible loss of data
C:\dev\wysiwyd\main\src\modules\attentionRelated\verbRec\src\verbRec.cpp(894): warning C4244: '+=' : conversion from 'float' to 'int', possible loss of data
C:\dev\wysiwyd\main\src\modules\attentionRelated\verbRec\src\verbRec.cpp(895): warning C4244: '+=' : conversion from 'float' to 'int', possible loss of data
```

You should replace `tabs` with `spaces` for a better indentation.

Cheers

pattacini, 2015-11-09:
Uhm ... your last commits caused history to mess up in GitHub @MagnusJohnsson 
Please always rebase before pushing ...

pattacini, 2015-11-10:
Hi @MagnusJohnsson 

Can you acknowledge you're receiving GitHub notifications?

MagnusJohnsson, 2015-11-10:
Acknowledged

pattacini, 2015-11-10:
:+1: Thanks

pattacini, 2015-11-12:
Hi @MagnusJohnsson 

I've done the required changes to fix the annoying warnings, please endorse them in your local repo.

Cheers,
Ugo

closed 2015-11-12 by pattacini

MagnusJohnsson, 2015-11-12:
Great!