robotology-legacy/wysiwyd #31: No tracking look

pattacini, 2015-11-09:
The `ICubClient::look()` remaps onto a **tracking** behavior of the gaze (by heritage); we should foresee a simple look service (i.e. no tracking) instead.

/cc @gregoire-pointeau

pattacini, 2015-11-09:
`ARE` sub-system is ready for that. Thus, the modification is to be applied to `attentionSelector` (where we have a `look(xyz)` but not a `look(object_name)`) and all the downstream modules need to be carefully checked/upgraded.

Tobias-Fischer, 2015-12-02:
@gregoire-pointeau @pattacini: Can we close this issue after our discussions yesterday? It seems tracking is what we actually want, so I don't see the need for the look method.

pattacini, 2015-12-02:
Fine for the `single look`, but I still have to check whether there'll be a weird interaction between `attentionSelector` and `ARE` during a commanded action (e.g. `point` ...).

Stay tuned!

Tobias-Fischer, 2015-12-04:
Actually, I think there is still some discussions to do here. Right now, both the `ARE` and `attention` subsystems offer a `look` service. In `iCubClient`, we employ the service from `attention` (which is then internally calling `ARE`). However, if `attentionSelector` is not running, this is not successful. What should the correct behavior be? Using `attentionSelector` by default, but if the subsystem is not connected fall back to `ARE`?

pattacini, 2015-12-04:
I can certainly implement as backup for `look` the call to `ARE` in case `attentionSelector` is not connected, but I don't think it will add that much. If we decided to go for the tracking look then I believe `attentionSelector` must be marked as mandatory component to run the demo.

pattacini, 2015-12-18:
What's the final decision on this?

DbD, 2015-12-18:
Ramping the project up again. Thanks Dick Woods 

Sent via the Samsung Galaxy S® 5 ACTIVE™, an AT&T 4G LTE smartphone

Tobias-Fischer, 2015-12-18:
We don't use attention selector here, so I personally think it would be nice to have a backup in that case :)
What do you think @gregoire-pointeau?

pattacini, 2015-12-18:
Ok then, I'll do the backup.

commit 46418255 (2015-12-19): [iCubClient]: look() calls ARE straightaway if Attention is not running; closes #31
closed 2015-12-19 by pattacini