robotology-legacy/wysiwyd #32: What's the status of iros2015 branch?

pattacini, 2015-11-12:
@jypuigbo @clement-moulin-frier

clement-moulin-frier, 2015-11-13:
Hi @pattacini 

We have to remove it from the remote repo indeed, we'll take care of this on Monday.

clement-moulin-frier, 2015-11-16:
Hi @pattacini ,

Actually we need a remote repo to synchronize the version we are working on between ourselves. We generally use a branch on robotology/wysiwyd to do so, before merging to master when the code is stable. Is it problematic?

Another solution is to fork the repo on one of our github account. Why not but branching is normally the way to work on development versions.

Thanks to let us know

pattacini, 2015-11-16:
No problem @clement-moulin-frier, keep using [**`iros2015`**](https://github.com/robotology/wysiwyd/tree/iros2015) for your convenience.

Tobias-Fischer, 2015-11-16:
Can I suggest renaming `iros2015` to something more expressive, like `reactiveLayer` or `upf-devel` or something like that? ;)

pattacini, 2015-11-16:
Good suggestion.

clement-moulin-frier, 2015-11-16:
Ok excellent idea, I've just done it. 
Side question before closing the issue: do we still have to be careful that the compilation works on other branches than `master`? (i.e will travis try to compile the non-master branches?)
Thx

clement-moulin-frier, 2015-11-16:
OK perfect

closed 2015-11-16 by clement-moulin-frier