robotology-legacy/wysiwyd #33: Repository quite big in size (120MB)

clement-moulin-frier, 2015-11-23:
Hi guys,

The repository is rather big (120MB), which made cloning quite a long process. I've made an analysis and it appears that files in `wysiwyd/main/app/visionRelated/headPoseEstimator/conf/trees/`  are taking around 40MB. This doesn't explain the 120MB but at least the main part of it.

Do these files really need to be versionized?

pattacini, 2015-11-23:
http://stackoverflow.com/questions/2116778/reduce-git-repository-size.

Tobias-Fischer, 2015-11-23:
1) Regarding the files for the headPoseEstimator: I'd prefer to have them versionized, otherwise people have to download them from somewhere else and place them in the appropriate directory if they want to run the module. 40MB at the end of the day isn't too much nowadays ..
2) Another explanation for the big size big be the following commit https://github.com/robotology/wysiwyd/commit/d795715dae700d2cdc4bcec2cbcfd597c38c1d93 and the revert of it https://github.com/robotology/wysiwyd/commit/3e7a239dc45b98f43bc340c38631935ac54283ff
I tried to get rid of them by rewriting history (scary!) but wasn't successful :(
If someone else wants to give it a try, that would be great.
3) Regarding the link from @pattacini, I am not sure whether it only reduces the size of the local repo or also of the global repo? If the latter, it would be great if you could give it a go, @clement-moulin-frier.

Best, Tobi

pattacini, 2015-11-23:
`git gc` acts on the local repository of course. Then, a `git push --force --all` is required with the dangerous caveat that the history will get modified and everyone is expected to perform a `rebase`.

Other details: http://stevelorek.com/how-to-shrink-a-git-repository.html.

Tobias-Fischer, 2015-11-23:
Scary! Let's give it a go during the integration meeting, where we also will be able to ensure that most people get a fresh copy of the repository after the compression will be done.

pattacini, 2015-11-24:
Indeed :smirk:

clement-moulin-frier, 2015-11-24:
Thanks for the info guys. I agree, it's better to check this during the meeting when everybody will be around. I close the issue for now.

closed 2015-11-24 by clement-moulin-frier

pattacini, 2015-11-24:
As side note, the git command to query the size: `git count-objects -vH`

The current result is:
```sh
count: 6772
size: 7.55 MiB
in-pack: 14022
packs: 14
size-pack: 118.56 MiB
prune-packable: 132
garbage: 0
size-garbage: 0 bytes
```

The interesting field is `size-pack`.