robotology-legacy/wysiwyd #34: Enable C++11 support for gcc

Tobias-Fischer, 2015-11-27:
Hi,
recently @gregoire-pointeau pushed some changes which require C++11 support. After some reading, I found that under Visual Studio the (partly) C++11 support is automatically enabled. However, when compiling with gcc, one needs to enable the C++11 support by setting the proper compiler flags.
I'd be happy to enable them, it's really easy. Then, we can go ahead with the beauty of C++11.

I'll do this in the integration meeting next week if no one has concerns. gcc now supports C++11 for a while, i.e. even gcc 4.6 shipped under Ubuntu 12.04 supports the core features.

Best,
Tobi

pattacini, 2015-11-27:
Fine with me.

Tobias-Fischer, 2015-11-28:
Fixed with https://github.com/robotology/wysiwyd/commit/50babc25a16ddb57aedbbc746a6d5e30fc5a0f99
I'll leave this open, as we should be able to get rid of the `boost` dependency in most cases now that C++11 is enabled by default. Currently, boost is needed for the following modules:
* ABM -> boost::thread can be replaced by std::thread
* reservoirHandler -> there is no equivalent to boost::python in standard C++
* SAM speechInteraction -> easy to get rid of boost::iequals using std::to_lower
* perspectiveTaking -> uses boost all over the place, however the new version of this module has a completely different structure anyways (using ROS ;) ), so I'll take care of this with the next update of the module

Tobias-Fischer, 2015-11-28:
Fixed with https://github.com/robotology/wysiwyd/commit/3d8c4efccc4dd2994b808abc91611eded135b0f4 and https://github.com/robotology/wysiwyd/commit/4ec27b10ee60912f7690d0c39d0926868494e196

closed 2015-11-28 by Tobias-Fischer

Tobias-Fischer, 2015-12-01:
To all Windows users: Please mind that this requires at least  Visual Studio 2012!