robotology-legacy/wysiwyd #38: Extend proactive tagging to be able to rename objects

Tobias-Fischer, 2015-12-07:
During the integration, we talked about having the possibility to re-train the names of objects. For example, the iCub might think an object is called `apple`, but it's actual name is `carrot`. This might be due to a) wrongly detected pointing in the training phase or b) `apple` might be a fantasy name of the iCub.

A simple implementation might be to just use the already existing code, but extend it such that objects which do not start with `unknown_object` are considered. Then, we should be able to say `iCub, point the carrot`. Then, iCub responds `I don't know which of these objects is the carrot`. We then point the `apple` (which is actually the `carrot`). The iCub then says: Now I know that the `carrot`.

A more advanced version would change the name step for step, so rather than one-shot learning we would need to teach the iCub the name of an objects over several iterations.

maxime-petit, 2015-12-08:
For this, there are 2 method in proactive:
- exploreUnknownEntity : does not care I think about the name unknown or not, so you can actually rename an object. It is just that the order (coming from allostatic) should allow that
- searchEntity : in that case, inside we check that "unkown" is part of the current name, so here we will need to change this

It is because the first one is for the robot self exploratory, and the second is the "goal-directed" exploration after an order. 

Basically, we may seen the "renaming" as a 3rd behavior and in that case take care of that independtly. Or we see that as a more "general" way to deal with the first 2 initial behaviors.

Thoughts?

Tobias-Fischer, 2016-03-17:
@clement-moulin-frier: What's the status on this?

clement-moulin-frier, 2016-03-17:
I've pushed a first version to the branch [upf/renaming](https://github.com/robotology/wysiwyd/tree/upf/renaming). I've also started to deal with random choice of sentences but can't really go further without being able to test it before.

Tobias-Fischer, 2016-03-20:
Hi @clement-moulin-frier,
Can you please open a pull request for the upf/renaming branch? This makes it easier to keep track of it. I hope we can test this soon and merge.

Best, Tobi

Tobias-Fischer, 2016-03-21:
We should also decide whether this is one of the features we want to show in year 2.
@clement-moulin-frier @pattacini @maxime-petit

clement-moulin-frier, 2016-03-23:
Hi guys,

I'll try and debug the new feature when I'll be back to the lab next week. I will make the pull request at this stage if the behavior is satisfactory.

Clément

Tobias-Fischer, 2016-03-31:
Any news on this @clement-moulin-frier?

clement-moulin-frier, 2016-04-01:
We were not able to finish it this week, we'll update next week through a pull request

Tobias-Fischer, 2016-04-07:
Hi @clement-moulin-frier, did you do further progress on this? I hope the refactoring did not cause too many issues.

clement-moulin-frier, 2016-04-07:
@jypuigbo is currently re-integrating the code after you pull request merge. It should be fine.

jypuigbo, 2016-04-11:
see PR #110

Tobias-Fischer, 2016-08-09:
Should we close this @jypuigbo @clement-moulin-frier @maxime-petit? Don't think this will happen, I think it's a `wontfix`.

clement-moulin-frier, 2016-08-11:
Yes let's close it for the time being and we'll reopen it if we need the feature for Y3. 
Not complicated to do but there many possible ways and we need to agree on one.

closed 2016-08-11 by clement-moulin-frier