robotology-legacy/wysiwyd #41: Ask to move bodypart

Tobias-Fischer, 2015-12-07:
We should be able to ask the iCub to move a bodypart, e.g. `iCub, move your index`. Then, the iCub responds `I don't know my index, can you please touch it`.

This extension is similar to the extension which was made at BCBT2015 for objects.

maxime-petit, 2015-12-08:
I can take care of this

./Max

Tobias-Fischer, 2016-01-07:
I have started tackling this issue in https://github.com/robotology/wysiwyd/commit/2782298e29990602aec92f5fb5562182866352e2

However, the following bits are missing:
1. `pasar` needs to be used to increase the saliency when the iCub is touched for the respective body part
2. the `iCubClient` needs a method to call the `babbling` module. If the name-joint association is already known, the robot will just move the corresponding joint
3. testing ..

I'll leave those points to you, @maxime-petit :)

Tobias-Fischer, 2016-01-07:
4. Add "Move your hand" etc. to `MainGrammar.xml` of `ears`

Tobias-Fischer, 2016-01-08:
@maxime-petit and I have found a fundamental problem here. After touching e.g. the iCub's index, the iCub will then know the relation between a set of taxels and a name, however the iCub is still missing the relation to the corresponding joint.
So either we have to assume this relation is known beforehand, or we have to somehow find the corresponding joint as well. @maxime-petit suggested that we could use the complaint mode to be able to move the joint, which would be quite neat, however I am not sure how easy this is.

Best,
Tobi

Tobias-Fischer, 2016-03-21:
@maxime-petit: Do you agree that this is work we will not show in year 2? If so, please change the milestone to year 3.

Thanks, Tobi

maxime-petit, 2016-03-23:
Hi,

If we start from a 100% unknwon bodypart (name = unknwon, tactile = unknown), has we said before, you can't ask the icub to move his index by just touching it (you miss the link between the tactile info and the name or joint).

However if the bodypart name is known we should be able to directly move it. In the same way, if we have the tactile info linked to a bodypart (for now the name is asked previously in any case,  but we can change that) you can touch the finger and the icub will be able to move it 

It is I guess linked to the "express knowledge" drives about bodypart, as @clement-moulin-frier presented it in #90.

I can do it, but I need to know where to put this (a new module? inside an already exisiting one that express knowledge about object, action, ...?).

@clement-moulin-frier @jypuigbo @sockchinglow

Tobias-Fischer, 2016-03-29:
@clement-moulin-frier @jypuigbo @sockchinglow: Any comments on this?

clement-moulin-frier, 2016-03-29:
Hi,

I think this behavior is rather related to the [`followingOrder`](https://github.com/robotology/wysiwyd/blob/master/main/src/modules/reactiveLayer/behaviorManager/src/followingOrder.cpp) behavior (when the human says "point to the X"). Looking at the code there is already a [`touchingOrder`](https://github.com/robotology/wysiwyd/blob/master/main/src/modules/reactiveLayer/behaviorManager/src/touchingOrder.cpp) behavior that is implemented, and the [main grammar](https://github.com/robotology/wysiwyd/blob/master/main/app/attentionRelated/ears/conf/MainGrammar.xml) is already able to deal with sentences like "Touch your thumb".

I guess this was done by Imperial, have you tested it?

Tobias-Fischer, 2016-04-07:
@maxime-petit: we should think about whether something is doable for the review.

maxime-petit, 2016-04-07:
I have added the new "babling" done when performing the tactile exploration, even if not trigger by a `followingOrder`, so it i clearly doable. https://github.com/robotology/wysiwyd/commit/46cdffca2642946d8fc675004a98a27fe8abc655

I ll put some code in `followingOrder` to trigger it when receiving `move bodypartname`. I will have that before the arrival in Lyon on the 13th.

./Max

maxime-petit, 2016-04-08:
Done with https://github.com/robotology/wysiwyd/commit/1152bd3835a15593b75c6391635d4c7b26dc6283

closed 2016-04-08 by maxime-petit

Tobias-Fischer, 2016-04-08:
Removed the old code in https://github.com/robotology/wysiwyd/commit/e6f7b44895f2c6ccc33719a95f8509747e21503e