robotology-legacy/wysiwyd #42: Run attention selector during proactive tagging

Tobias-Fischer, 2015-12-07:
Attention selector should run in the demo, so the iCub appears more alive .. :).

This comes with the problem that some objects might disappear from the view.

maxime-petit, 2015-12-08:
@Inserm the FPS for the IOL is 10 FPS, so it might not be so much of a big issue under "normal" GPU CUDA optimisation for the "smoothness". 

However, we have to implement a small "searching" behavior if the robot is just looking to 1/2 object because the 3rd one is none visible when looking the others

Tobias-Fischer, 2015-12-08:
I guess it would be a good idea if someone from a lab with the cluster is taking care of this, so we have similar conditions to what we will have in Lyon.

clement-moulin-frier, 2016-02-22:
We are currently running Attention Selector during the proactive demo and it works quite well (the robot indeed appears much more alive).  The only problem is that when pointing at objects, the robot can sometime look at another one. What would be the best way to focus the attention of the robot on the object it is pointing? By increasing its saliency through Pasar? If so, is Attention Selector able to read the saliency values set by Pasar to prioritize where it is looking?

PS: Note that we are currently working with a pre-populated OPC, so we do not run IOL and colleagues.

pattacini, 2016-02-23:
Hi @clement-moulin-frier 

Actually, if the `point` is performed via `ARE`, the handling of attention should be already taken into account: see [[1]] and [[2]].

Could you check that the connection between the two sub-system is in place?

[1]: https://github.com/robotology/wysiwyd/blob/master/main%2Fsrc%2Flibraries%2Fwrdac%2Finclude%2Fwrdac%2Fsubsystems%2FsubSystem_ARE.h#L329
[2]: https://github.com/robotology/wysiwyd/blob/master/main%2Fsrc%2Flibraries%2Fwrdac%2Finclude%2Fwrdac%2Fsubsystems%2FsubSystem_ARE.h#L67

gregoire-pointeau, 2016-02-23:
However, increasing the salience of the pointed object is mandatory I would say: if you point to an object, its salience should increase.
However you don't need to go through `pasar` to do it, a simple `opcClient` is enough.
Maybe in the `susystem_ARE` we could had to increase the salience of the pointed object. I'll look at it.
(increase the salience is needed independently from the problem of the `attentionSelector`)

pattacini, 2016-02-23:
@gregoire-pointeau don't mix things: `SubSystem_ARE` is for controlling the robot movements, so it's a very very ... very very bad location where to deal with object salience ;)

pattacini, 2016-02-23:
Salience handling should be running somewhere in our architecture with the aim to provide the robot with some points of interest (POI) where to focus the attention.

Once a POI gets selected, `SubSystem_ARE` kicks in and should prevent `attentionSelector` to redirect the gaze elsewhere for the entire duration of the point action.

gregoire-pointeau, 2016-02-23:
I disagree a bit. `ARE` is for controlling the robot, where the `subsystem_ARE` is just something to link `ARE` and the `WR` environment.
This is why for me the `subsystems` are for the `WR` related.
I don't see why the `subsystem_ARE` couldn't handle the salience.

pattacini, 2016-02-23:
`SubSystem_ARE` is a mere interface to `ARE`, so please don't put anything related to salience therein: it's really against software engineering rules.

The salience of an object has nothing to do with the robot movements: let's not mix inputs to the motor control with motor control itself.

maxime-petit, 2016-02-23:
Gregoire you said that if you point to an object, its salience should increase, that is how you motivate `ARE` or `subsystem_ARE`to be involved in the saliency. But for me it is actually the opposite: you point an object because the saliency is already high. The saliency cause your pointing, so the saliency should be high BEFORE you do the pointing: that is why you point this precise object and not another one. The pointing of another agent however, cause the saliency.

In proactive, I guess what could happen is that the robot can only point to an object with unknown name even if the actual highest saliency (and thus the gaze) is for another object. So without any goal, or with the goal of learning, the saliency should be increase to unknown object for instance?  

Another solution as you suggested Gregoire can be to change directly the saliency of an object when you have a precise behavior, but in that case it is similar to Ugo proposal to have a module "taking over" or "ignoring" the saliency when you have to do something specific. It is more clean in a coding way (the only module that change the saliency is the module that is actually suppose/design/responsible for the saliency), maybe a bit more weird in a "cognitive" way. An hybrid solution might be to have the module sending a command to `pasar` to increase greatly the saliency in an instant manner but this can not guarantee I guess an 100% "succes/consistent" gazing.

stephane-lallee, 2016-02-23:
In the view I had back in the days, increasing this saliency based on a top-down symbolic information (an object is pointed at) would have been the role of Pasar.
Of course I do agree with Ugo that putting saliency information in the ARE subsystem is a terrible idea 😉

From a more conceptual point of view, isn't there something wrong in increasing the saliency of an object you are pointing at? If somebody else is pointing I agree but if it is yourself... I would say there is there some process similar to the inhibition of return that would actually decrease the saliency of the pointed object.

My suggestion would be : Point at the object, but loon at the agent to who you are showing the object. Because pointing is a communicative act...

And pointers are great. (was I not cleaning properly ?😉  )

-----Message d'origine-----

clement-moulin-frier, 2016-02-23:
We are going to try this:
In `proactivetagging`, increase the saliency of the pointed object as well as the human agent, so that the robot gaze switch between both (like an infant would do actually). Can you confirm that we have to use Pasar for this? Or can we mor simply modify the saliency value using OPCClient?

Another question is: once the saliencies are modified, are they taken into account by Attention Selector? (i.e. does Attention Selector preferably look at the most salient object, or does it simply choose objects to look at them randomly?). 

@pattacini : can you please precise what connection we have to check?

Thx

gregoire-pointeau, 2016-02-23:
You don't need to go through `pasar` to change the salience. `pasar` will change automatically according to the acceleration, pointing, waving and appearance disapperance.
Can you check that `pasar` works correctly? (go in front of the iCub with `agentDetector`, move and the salience of the agent should rise).

Then if you artificially increase the salience of an object, with `pasar` running it should decrease, but keep still if not.

clement-moulin-frier, 2016-02-23:
Thanks. One problem is that even without Pasar running, the saliency is rapidly reset to 0 when the robot is pointing. Besides Pasar, do you know if there are other modules able to modify the saliency?

gregoire-pointeau, 2016-02-23:
Any module that access the `OPC` can modify the salience. But I don't know other than `PASAR` that does.

pattacini, 2016-02-24:
@clement-moulin-frier you could check that the following functions are actually called by putting some printouts within the code:
- [**`SubATT->stop()`**](https://github.com/robotology/wysiwyd/blob/master/main/src/libraries/wrdac/include/wrdac/subsystems/subSystem_ARE.h#L76)
- [**`SubATT->track()`**](https://github.com/robotology/wysiwyd/blob/master/main/src/libraries/wrdac/include/wrdac/subsystems/subSystem_ARE.h#L88)

Tobias-Fischer, 2016-02-29:
Hi @clement-moulin-frier,
Have you further debugged this to see whether the attention system is called? We should figure this out before the integration meeting.

Best, Tobi

clement-moulin-frier, 2016-03-04:
Hi @Tobias-Fischer 

We haven't figured out what is the reason of the saliency modification, see my last but one message (the long one).

Tobias-Fischer, 2016-03-16:
Thanks to @pattacini and https://github.com/robotology/wysiwyd/pull/82 we got much closer to this goal. I suggest we sit together today and look for the reason why the saliency is reset to 0.0 as reported by @clement-moulin-frier.

I suggest we run the proactive tagging demo with `attentionSelector` and see how it goes.

Also @gregoire-pointeau should change `pasar` because of the slight position changes due to the vision changes, which lead to a too high saliency right now.

jypuigbo, 2016-03-19:
Did this meeting take place? Which is the current status of the issue, after the integration? Thanks

Tobias-Fischer, 2016-03-21:
From the debriefing email:
```
Progress towards running the attention selector during the demo was made. Objects now don't disappear when the iCub is looking to another object or the partner, previously leading the object to be set as not present in the OPC, as it was out of the field of view. However, we decided that for the year 2 demo we will not run attention selector, and do final tests before the year 3 review.
```

Thus, taking the high priority tag off. Let me know if you think otherwise @pattacini @gregoire-pointeau

gregoire-pointeau, 2016-03-22:
Going from a boolean to 0 - 0.5 - 1 induce indeed many changes in the higher modules that for now are not taken into account and I agree we need more time to have some behaviors coherent with it.

Tobias-Fischer, 2016-03-22:
Normal modules shouldn't check for the 0.5 value. That value is only meant to be used in `iol2opc` and `attentionSelector`, and @pattacini already took care of that.
However, I agree that there is now a difference of checking for == 1.0 vs != 1.0 vs == 0.0 which might have introduced new bugs we have to check again.

gregoire-pointeau, 2016-03-22:
I don't totally agree with the fact that modules should check for 0.5, it depends of the behavior you want to have and this needs more time to think about that just a few weeks. For example in term of narrative, this induces many changes, and also in term of actions consequences for abmReasoning...
I agree however that it is an huge and useful improvement, but not as trivial at is seems for "higher" modules.

pattacini, 2016-03-22:
@gregoire-pointeau when I introduced this change I payed particular attention to retain the old intended behavior in all the modules involved.

It should be straightforward to go through an exhaustive search for the flag `m_present` and do a double check.

pattacini, 2016-03-22:
To recap:

The old check
```cpp
if (m_present)
```
should become
```cpp
if (m_present==1.0)
```

The old check
```cpp
if (!m_present)
```
should become
```cpp
if (m_present!=1.0)
```

gregoire-pointeau, 2016-03-22:
Yes indeed in term of computation, it doesn't change much I agree, but if we want to take these changes into account and to use it, it won't be that trivial.
But in term of retro compatibility, I guess it doesn't change much indeed.

pattacini, 2016-03-22:
I agree that for now we shouldn't employ this new piece of information for new devs.

Tobias-Fischer, 2016-06-07:
@clement-moulin-frier @gregoire-pointeau @maxime-petit @pattacini 
Do you think this should be tackled in year 3?

pattacini, 2016-06-07:
I have no objections to try it out again for year 3.

maxime-petit, 2016-06-07:
Yes sure,

Maybe it does not have to be permanent but at least in some "subset" of scenario (like an active search, ...)

clement-moulin-frier, 2016-06-07:
Would be a very good feature to have indeed. But we (upf) still haven't understood why and how the saliency decreases automatically (and very rapidly), preventing us to control it programatically in a robust way.
Does the new convention for `m_present`, now taking real values, change something?

Tobias-Fischer, 2016-09-16:
I don't think anyone is planning to show this. Closing.

commit 428c2c83 (2016-04-15): [ABM] Fix #42
closed 2016-09-16 by Tobias-Fischer