robotology-legacy/wysiwyd #43: Demo with two different agents and different set of objects

Tobias-Fischer, 2015-12-07:
@gregoire-pointeau was saying that we should extend the demo so the iCub is interacting with two different agents, and each agent might use different objects.

gregoire-pointeau, 2016-01-19:
Concerning the different objects, there are no problem.
Concerning the agent, maybe we can put in `pasar` that a disappearance of an agent, puts his/her name to "partner" or "unknown" again ?

Tobias-Fischer, 2016-01-19:
I think `agentDetector` used to deal with different agents. Internally, the kinect already can differentiate different agents somehow. We lost this ability when we changed `agentDetector` such that the name of an agent can be changed. However the code hasn't changed much, and we should be able to go back to the old behavior.

Tobias-Fischer, 2016-02-29:
This needs proper handling of https://github.com/robotology/wysiwyd/issues/51 first!

Tobias-Fischer, 2016-03-21:
@gregoire-pointeau: Do you think we should work on this for the year 2 or year 3 demo?

gregoire-pointeau, 2016-03-22:
The idea to bring new objects is still interesting for me. For example, a first set of `proactiveTagging` with 2 or 3 unknown objects, then we introduce new unknown objects. But to be able to do the demo on more than 3 objects present all the time.
Concerning the agent it's a bit more tricky while we don't have the face recognition running and integrated.
So my idea whould be: objects Y2, agents, Y3.

Tobias-Fischer, 2016-03-31:
Hi @gregoire-pointeau,
As far as I can tell multiple sets of objects are already working. You just need to run `iol2opc` so that all objects are detected as `unknown_object*`, and then you can remove some of them and introduce them later?!?

gregoire-pointeau, 2016-04-01:
Yes indeed it works perfectly with as many objects as we want. Especially is the `attentionSelector` is not running.
We just need to find a way to introduce objects in the demo smoothly but this is just a matter of scenario.
As I said, the tricky part is only for the agents.

Tobias-Fischer, 2016-04-27:
Closing, as https://github.com/robotology/wysiwyd/issues/51 covers different agents, and different set of objects is already possible.

closed 2016-04-27 by Tobias-Fischer