robotology-legacy/wysiwyd #44: Integrate perspective taking for looking to find out which object human is looking at

Tobias-Fischer, 2015-12-07:
Rather than just using the pointing action, we should be able to use perspective taking to infer which object the human is looking at. This can be used to increase the salience of an object, and thus for the proactive tagging demo.

Tobias-Fischer, 2016-08-17:
If we decide that this is needed in the scenario , the first step to tackle this issue would be making the output of the kinect wrapper compatible with `pcl`. 

- So far there is only a method to retrieve single 3D points using a RPC command, which will be too slow to reconstruct the point cloud. 
- Instead, I guess it will be faster to either construct the point cloud within `kinect-wrapper` and send the data via a `BufferedPort` (using e.g. http://docs.ros.org/api/sensor_msgs/html/msg/PointCloud2.html). 
- The other option is to do the reconstruction within the perspective taking module using the depth image (RGB information is not really needed).

What do you think @pattacini? But as I said, before doing any work on this we should be 100% certain that we actually need this in the scenario.

pattacini, 2016-08-17:
Hi @Tobias-Fischer 

I think `kinect-wrapper` could be conveniently extended in order to expose a new method (say `get3DPoints`) responsible for providing the entire point cloud extracted from the image. That being said, I'd rather keep the library independent from `PCL`, which is really not required to this end.

Then, copying out the output of `get3DPoints` in a `PCL` compatible object would likely introduce a latency, but it's a price we can pay to guarantee to be safe against the _dependency hell_ :smile:

Tobias-Fischer, 2016-09-16:
Closing this issue, it's been integrated. We'll see how to demonstrate it in the demo.

closed 2016-09-16 by Tobias-Fischer