robotology-legacy/wysiwyd #45: Proactively tag actions

Tobias-Fischer, 2015-12-07:
In the meeting with Yiannis, he suggested that one possible way to extend the demo might be to proactively tag actions. I.e. the iCub should be able to discover the names of the actions he is able to do. We should start with looking, pointing and pushing.

gregoire-pointeau, 2016-01-04:
To be able to do so, it would be nice to have a set of unknown action to perform (maybe it's already in proagtiveTagging.cpp ?).
Another issue is to have a "dictionary" of actions I guess.

Tobias-Fischer, 2016-03-21:
Are we going to show this in year 2? If not, we should take the high priority label off.
@clement-moulin-frier @maxime-petit @gregoire-pointeau @pattacini

gregoire-pointeau, 2016-03-22:
I guess for the demo Y2, it's too late.
I think we should however keep it for Y3.

clement-moulin-frier, 2016-03-23:
I agree with @gregoire-pointeau , unless someone has a clear idea about how to do it in a very simple way.

Tobias-Fischer, 2016-08-09:
Hi @clement-moulin-frier @gregoire-pointeau, should we close this? Or is anyone working on this?

clement-moulin-frier, 2016-08-11:
It was mentioned during the videoconf workshop that proactive tagging could focus more on actions for Y3 (in order both to keep it and not re-presenting the exact the same thing).

@dcam0050 Do you think you could have a look at it during BCBT? As mentioned during the workshop it can be quite easy if we simply consider tagging actions that the iCub is already able to classify (i.e. to distinguish as unknown_1, unknown_2 using a pre-trained model).

Tobias-Fischer, 2016-10-27:
@dcam0050 did you work on this?
@hjchang @maxime-petit can we use the "abnormality" detection of object use we discussed the other day?

Tobias-Fischer, 2017-01-05:
Closing this as no one seems to work on this.
/cc @hjchang @dcam0050 @clement-moulin-frier

closed 2017-01-05 by Tobias-Fischer