robotology-legacy/wysiwyd #46: Integrate reservoir for language understanding and production

Tobias-Fischer, 2015-12-07:
See title

gregoire-pointeau, 2016-01-18:
A first version of the subsystem if created.
Coming next, in "WYSIWYD2016 Resurrection", the proper tools to install `LRH`

gregoire-pointeau, 2016-01-29:
`lrh` has now a `subsystem` with 2 functions: `meaningToSentence` `sentenceToMeaning`.
For now the `meaning`, the formalism is still a bit obscure.
I'll try to have something easier to use.

Tobias-Fischer, 2016-02-29:
Hi @gregoire-pointeau and @AnneLaureM,

Did you guys work further on this? Is it ready for the demo?

Best, Tobi

gregoire-pointeau, 2016-03-07:
Hi Tobi,
NarrativeHandler is currently using `lrh` for the demo.
However the version we use it quite slow. Running but slow. We hope to fix it before the integration.
The `subsystem_LRH` can be used as long as you have `LRH` running.
There is 2 functions: `sentence_to_meaning` and `meaning_to_sentence`.
However I admit that the format of meaning we use currently is a bit complex, we are working on changing it for something easier.

Tobias-Fischer, 2016-03-21:
Hi @gregoire-pointeau @AnneLaureM,
What is the status of this? Are we still planning to show this in year 2?

Best, Tobi

gregoire-pointeau, 2016-03-22:
Yes, the narrative part where the robot is boring, will be boring but with `lrh`.
The idea is that the robot will first create a simple narration (infant one) using `lrh` in production, then will ask the human to tell the story, and will understand the human through `lrh`.
For now it's working fine, we only have to change the way we do our learning to speed up the speech production.

AnneLaureM, 2016-03-23:
Hi @Tobias-Fischer, I will take care of it right away.

Tobias-Fischer, 2016-03-31:
Any news on this @AnneLaureM ? Will it be ready for year 2 or shall we tackle this in year 3? If we want to use this, I guess a lot of changes all over the place need to be done ...

gregoire-pointeau, 2016-04-01:
Hi @Tobias-Fischer ,
concerning the integration of the `lrh` and narrative in the demo, we came up with an idea.
The full plan that we had originally was the following:
1 robot and human act
2 robot express its naive narrative (like we have seen during the IM)
3 human gives the robot a human-like narrative
4 robot uses the human narrative

But all of this would be too long.
So we decided to do all of it the week before, and during the demo, the behavior `narrate` will trigger only the 4th part, where the robot will express the narration of a demo we would have record the week before using `lrh` and `narrativeHandler`.

And the demonstration of the full system will be shown during the presentation with video that we recorded here on shorter examples.

Tobias-Fischer, 2016-04-01:
Hi @gregoire-pointeau,
Sounds good to me! Is this already working (at least as prototype)?

Best,
Tobi

gregoire-pointeau, 2016-04-01:
Well, most of it.
The only thing we need to had, it to reload the human narration (until now we were focusing on creating it), then we will need to find a good demo, and get the `instance` in the ABM to access it easily.
Like this the "macro" scenario should be:
- trigger narrate
- iCub : "Remember that yesterday, I wanted to know what was an object, and you told me it was the croco, then you asked me to point as the rabbit, but I didn't know what was the rabbit so you showed me the rabbit..."

It should ask a bit of development, but only to store and retrieve sentences in `ABM`.
I guess it'll be finished by the end of the day.

gregoire-pointeau, 2016-04-01:
Hi. This is done.
What remain to do, ti to record a full interaction with the robot similar to the one we will do for the review, then give the iCub the corresponding narration that will be analyzed by `lrh` and record in ABM, and note the instance of this story in `narrativeHandler.ini`.
Then, the `narrate()` function call the narration given previously.

Tobias-Fischer, 2016-04-01:
Well done. Sounds good.

gregoire-pointeau, 2016-04-07:
Last commit uses every abilities of both `lrh` and `narrativeHandler`.

closed 2016-04-07 by gregoire-pointeau