robotology-legacy/wysiwyd #47: Use narrative handler

Tobias-Fischer, 2015-12-07:
@gregoire-pointeau suggested to use the narrative handler in the proactive tagging demo, so the iCub can talk about past events. Please add some more details what you imagine @gregoire-pointeau

maxime-petit, 2015-12-08:
We can try also to use the narrative module as soon as it is pushed/documented, in order to compare between INSERM/IMPERIAL ABM.

gregoire-pointeau, 2016-01-29:
first version is in place.
People with a `ABM` can try to launch `narrativeHandler` by playing with the parameters in `narrativeHandler.ini`:
- `dThresholdDiffStory` is the timing between two event from a same story
- `dThresholdSizeStory` is the size minimum of a story in term of events.
- `lrh` should be at `1` if you want to use `lrh` to create the story, either only a simple `A-P-O-R` form is applied
- `forwardABM` decide if the new story created should populate also the ABM (risk of infinite story + narration in loop), better set to `0`, WIP.
- `instanceStart`   you can decide when to start looking for stories in the `ABM`

Tobias-Fischer, 2016-02-29:
Hi @gregoire-pointeau,
What is the status of the narrative handler? Is it ready to be used in the proactive tagging demo?

Best, Tobi

gregoire-pointeau, 2016-03-07:
Hi Tobi,
NarrativeHandler is working as a prototype version properly.
For now the story are a bit "rude" but it's improving.

I still have a few changes to make here and there especially to record the good name of agent when using `tts` or `stt` (for now it's always agent). And also I have a few problem with fluctuation of appearance and dissapearance of object (which create strange stories).

closed 2016-03-07 by gregoire-pointeau