robotology-legacy/wysiwyd #48: Change pasar to be able to deal with names different than "partner"

Tobias-Fischer, 2015-12-07:
Right now, pasar is relying on an agent called `partner` for the pointing detection etc. - this needs to be extended.

Anyone from UPF willing to take care of this? @jypuigbo @clement-moulin-frier @sockchinglow

Tobias-Fischer, 2015-12-15:
This can be easily done using the commit for agentDetector (see https://github.com/robotology/wysiwyd/issues/51 and https://github.com/robotology/wysiwyd/commit/fb464122a4bfff79f04c96910136bb6615032916) as template. Someone please assign this issue to him/herself; let me know if you have questions.
@sockchinglow: This might be a good opportunity to get familiar with some of the core systems of WYSIWYD.

Best, Tobi

gregoire-pointeau, 2016-01-15:
The kinect cannot detect more than one agent.
I would suggest to replace: `name = "partner"` by `isAgent()` and `name != "iCub"`

Tobias-Fischer, 2016-01-15:
Ahhh I didn't know that. Works for me then!

Tobias-Fischer, 2016-01-15:
Maybe `name!=iCub && name.doesnotstartwith("unknown") && isAgent()`?!?

Thanks for pointing this out @maxime-petit

gregoire-pointeau, 2016-01-15:
Well it depends if we can have several agents in the opc. Normally yes (iCub + partner).
But kinect is only able to detect once at the time.
So if there is an agent which is not the iCub, it should be the partner no ?

Tobias-Fischer, 2016-01-15:
Ahh yeah, for the pointing context it does not matter what the name of the partner is. We should still check the `isPresent` flag though.

Best,
Tobi

gregoire-pointeau, 2016-01-15:
Indeed. So for pasar, I don't need to check that the name is not `unknown`, because an unknown agent can move or point anyway right ?

Tobias-Fischer, 2016-01-15:
Yes, correct.

maxime-petit, 2016-01-15:
In order to cope with potential improvements (e.g. buying a Kinect2 that allow multi-users skeleton) if you can rise a warning if there are more than one agent (not iCub) present (even if in that case you take the first one by default or something else)

It can also be used in that way for catching potential error of updating the presence of agents

gregoire-pointeau, 2016-01-18:
Done.

closed 2016-01-18 by gregoire-pointeau