robotology-legacy/wysiwyd #50: Reduce speech recognizer time delay

Tobias-Fischer, 2015-12-07:
So far, we need to wait several seconds before a sentence is recognized when the grammar of `speechRecognizer` is changed. I think a clean solution would be to interrupt the loop in which the `speechReognizer` is stuck when called with `recognizeGrammerLoop` (or something like that), so a new grammar can be loaded faster. That seems to be cleaner than reducing the time of the loop.

@maxime-petit: Let's tackle this together.

pattacini, 2015-12-07:
/cc @robotology/speech-developers

stephane-lallee, 2015-12-07:
As far as I remember the length of the recognition timeout was a parameter of the SAPI method. I'm quite confident that you cannot interrupt "cleanly" the loop as there was no asynchronous/callback based method. But feel free to try and please keep me posted if you manage to sort it out.

If you do then that would also solve the "hanging" problem when you close the module with ctrl-c.

-----Message d'origine-----

maxime-petit, 2015-12-11:
Yop!

We should not wait for the iCub to ask a question/sentence before sending the grammar to be recognized, in that case we can "earn" 1 second. I have also reduced the delay in the .ini file of the speech to 1000 ms (in my own local .ini file so you have to do the same). It is fine because the system blocks the new grammar when you are currently speaking, so it does not matter if you talk for more than 1 second :)

Also, be sure to mute/unmute your microphone. You need to have the line:
'========================== Go Wainting for recog! =============='

This tells you that the grammar is properly loaded. If you speak before that (e.g. your microphone is on and you have some background noise) the recognition will not reply to the module which has loaded the grammar. Instead, you can see a list of all of the word recognize preceeded with [Debug]. 

It was working fine with bodypart naming. I ll check next week with object and agent tagging, and potentially to change the speech per see, in case there is an interrupt method in SAPI.

Tobias-Fischer, 2016-01-07:
Hi @maxime-petit,

I remember we had worked on this before the Christmas break. I think we can close this issue.
We just have to be careful always first loading the new grammar, and afterwards use e.g. `iCub->say()`, `iCub->point()` etc. This way, the grammar is changed before the iCub is asking for a response, and thus after the `iCub->say()` the new grammar should be loaded in most of the cases.

Please correct me if I am wrong :).

Best, Tobi

Tobias-Fischer, 2016-01-27:
See https://github.com/robotology/wysiwyd/commit/bb849b72f5d7cf1ac35f3623e090176eecbd7cf6

Tobias-Fischer, 2016-02-29:
Hi @maxime-petit,
Can we close here?

Best, Tobi

maxime-petit, 2016-03-01:
Yes I think so.

closed 2016-03-01 by Tobias-Fischer