robotology-legacy/wysiwyd #51: agentDetector should not detect unknown partner

Tobias-Fischer, 2015-12-07:
We had the issue in the integration meeting that `agentDetector` always detected an agent called `unknown` which is present, although there is another agent called `partner` which is the agent we actually want to recognize.
Further, I suggest to rename `unknown` to `unknown_agent`.

stephane-lallee, 2015-12-07:
In fact there was a piece of code commented out within agent detector that was proceeding to face identification if the correct module was present. Then the skeleton would be named a d tracked properly. We never really used it, but if you have time to spend it is a required step toward the "ultimate child robot" ;-)

-----Message d'origine-----

Tobias-Fischer, 2015-12-10:
@maxime-petit and I tackled this today, and will commit the updated version after some more tests. However, the drawback is that only one partner is allowed, so multiple skeletons will not show up. I guess we need to find a way to handle multiple simultaneous partners.

Tobias-Fischer, 2015-12-15:
This is now working nicely, however still under the assumption that only one partner is in the scene. Also, `agentDetector` can now handle a partner named differently than `partner`, and the `iCubClient` automatically informs `agentDetector` if the name of an agent was changed (`changeName` method). Obviously, this needs to be extended, but I guess it's a good start.

Tobias-Fischer, 2016-03-21:
Hi @gregoire-pointeau @maxime-petit @pattacini @clement-moulin-frier,
Do you think we need this for the year 2 demo? I guess it's good enough if we fix this for the year 3 demo.

Best, Tobi

gregoire-pointeau, 2016-03-22:
See issues #43 for my opinion on different agents.

Tobias-Fischer, 2016-05-09:
@dcam0050 are you planning to get face recognition running on the iCub (within WYSIWYD)? This would fix this problem. Look at the boolean variable `useFaceRecognition` and where it is used. Also, check `handleMultiplePlayers`.

In general, I think this issue would be best tackled at an integration meeting. Before tackling it, we should be very sure though that we end up using it. I am guessing it will easily take a full day before it's working nicely.

Tobias-Fischer, 2016-08-09:
Hi @dcam0050 and @gregoire-pointeau,
I don't see this happening in the next few months, should we close this as `wontfix`?

Tobias-Fischer, 2016-10-03:
Closing as `wontfix`, let be know if there are any objections.

closed 2016-10-03 by Tobias-Fischer