robotology-legacy/wysiwyd #58: Populate object names to the different grammars

Tobias-Fischer, 2015-12-10:
Right now, both grammars of `proactiveTagging` as well as `ears` need to be changed when adding/removing/changing the names of objects. This should be somehow unified.

@maxime-petit suggested to retrieve the possible names of objects from the `autobiographicalMemory`.

Best, Tobi

gregoire-pointeau, 2015-12-10:
One issue with it, would be that for the proactiveTagging scenario from a very pragmatic point of view, we would always say: "I know what is a croco, and the croco is not here".
So, maybe a boolean to populate or not this vocabulary would be good I guess.

Tobias-Fischer, 2015-12-10:
We are only talking about the grammar of the speech recognizer here. This should not change any behavior of any of our modules.

Tobias-Fischer, 2016-05-09:
To be honest, I don't think anyone will ever implement this. If someone will look into it, the place to do so is `grammarToString` (wrdac/functions.cpp). There, one could either implement loading a XML within a XML, or loading objects from the ABM and creating some XML code from that.

Closing here, if you think it is important feel free to re-open @maxime-petit.

closed 2016-05-09 by Tobias-Fischer