robotology-legacy/wysiwyd #60: [iol2opc] Add options --empty and --no_save

Tobias-Fischer, 2015-12-15:
With similar behavior to the `objectsPropertiesCollector`.

This is a low priority issue.

pattacini, 2015-12-20:
I've added the `--empty` option to `iol2opc` and it should work.

Conversely, the `--nosave` is tricky 'cause cannot be localized in just one place. As soon as the user stops the application, the modules can shut down independently and in a random order, thus not leaving any room for communicating that the `--nosave` process needs to be done.

I really don't like to spread this options among multiple modules and, further, I think it is not necessary in the end. Launching `iol2opc --empty` should be sufficient, since it's true that the database will be saved upon all the operations but then it will be cleared at start-up.

Tobias-Fischer, 2015-12-22:
Great, thanks a lot Ugo! I didn't think about the issues regarding `nosave` and agree it would be too much hassle to implement.
I'll close this issue as soon as I tested the functionality of `empty` (probably in the new year).

Thanks, Tobi

Tobias-Fischer, 2016-01-04:
Working perfectly, thanks!

closed 2016-01-04 by Tobias-Fischer