robotology-legacy/wysiwyd #62: Visualization of drives

Tobias-Fischer, 2016-01-04:
Hi @clement-moulin-frier, @jypuigbo and @sockchinglow,

according to @maxime-petit there was a visualization of the drives, their comfort-zones, and the prioritization between drives in EFAA. It would be nice to have this kind of visualization also in WYSIWYD, as 1) it simplifies debugging purposes, to see why the current drive was activated / when the next drive will be activated etc. and 2) the proactive tagging demo then makes more sense, as one can explain what is going on much easier.
According to Max, this was previously part of the `iCubGui`. I am not sure whether it will be easier to match your new pipeline to the existing graphical tools, or whether it's faster to write a Python script which does the job, or maybe even `yarpscope` does the job.

@clement-moulin-frier: Assigning you to do the job, please re-assign if someone else takes over ;).

Best,
Tobi

stephane-lallee, 2016-01-04:
Yes, those were simply colored boxes in iCubGui. I also did a better visualization in C# at some point but the code is lost.

My suggestion would be for you to go for some python gui as there is no proper way to display 2D element in iCubGui. For yarpscope it would require writing a small module retriving the drive values and streaming them as a single bottle.

-----Message d'origine-----

clement-moulin-frier, 2016-01-05:
Yes I agree it is an important feature to have. @jypuigbo from what I remember you have already coded something in Python to do this (ethogram)?

maxime-petit, 2016-01-05:
We might also use the same kind of histogram that you can have in iol (drawScoresHistogram method?), with the objects names probability? In that case we keep it yarpview/c++ and we can reuse the code

./Max

pattacini, 2016-01-05:
Good idea @maxime-petit

clement-moulin-frier, 2016-01-19:
The problem in using an histogram is that we loose the evolution of drives through time. I would rather go for a graph with time on the x-axis and drive values on the y-axis (better for explaining during the review, and maybe also for debugging). Do you agree?

I can do it using yarp view if there is a way to label the plots (like a colored legend, or the name of each drive next to its curve). Do you know if it is possible in yarpview?

Otherwise I can also do it in Python (probably much easier for me).

maxime-petit, 2016-01-19:
Actually I think both is better :D 

The plot for some history, the histogram to show better (because bigger and with the label) what is the biggest current drive (and so the one that is currently triggering the behavior) maybe with the threshold visible?

The histogram thing was also because it is already implemented for the object name so it should be too long to use it for drives? The plotting things might take more time (just a guess depend on your confidence in this area). Also if you use Python I would say that if it is embedded fine, but I would be against if it add another model (that might need to run by hand and not through gyarpmanager)

I would rather have something roughly ok "for tomorrow" and then have room for improvement :) If you are confident and that does not add complexity in term of running the whole thing I am fine anyway

./Max

clement-moulin-frier, 2016-01-19:
OK, to have something ready asap, I prefer to do it in Python (I have no experience with visualization in C++ or with yarp).
What do you mean by " if it is embedded fine"?

maxime-petit, 2016-01-19:
I mean we should NOT run the python script or another additional module for visualization. It should be running when we just launch homeostatis/allostatic :)

Tobias-Fischer, 2016-01-19:
I would be fine with another (python) script to run for now. As long as it's installed in the proper folder, we will be able to use yarpmanager to launch it, so I think it's fine.

clement-moulin-frier, 2016-01-19:
mmm, not sure about it @maxime-petit . E.g. you might want to run the module on a computer with no graphics (what would make the module crash if visualization is directly embedded in it)

clement-moulin-frier, 2016-01-19:
OK I will go for a python script as a first trial, then we'll see. I try to figure this out this afternoon.

maxime-petit, 2016-01-19:
Yep sure :)

clement-moulin-frier, 2016-01-20:
Hi there,

I've a first version ready to test. However it is not clear how to set cmake so that the script is installed properly. Anyone has an experience with this?

Thx

Tobias-Fischer, 2016-01-20:
Hi,
have a look at https://github.com/robotology/wysiwyd/blob/master/main/src/modules/samRelated/SAM/SAM_Demos/CMakeLists.txt

Looking forward to the commit!

Best, Tobi

clement-moulin-frier, 2016-01-20:
Thanks @Tobias-Fischer 

I have pushed it, with update in the cmake to install it correctly. It still have to be improved a lot but please first try this version (see below for how to do) and give me your feedback about what features/improvements would be useful (see below about the one I plan to do).

## To run it:
You have to run [`allostatic_plot.py`](https://github.com/robotology/wysiwyd/blob/master/main/src/modules/reactiveLayer/allostatic_plot.py) whenever you want after having launched `homeostasis`. 
X-axis is time (on a sliding time window of 20s), y-axis is the drive level. Horizontal dashed lines are the homeostasis boundaries (comfort zone). Thick lines are the (evolving) drive levels. Each drive corresponds to a color as indicated in the legend (labels are the name of the drives as defined in the [`allostaticController` config file](https://github.com/robotology/wysiwyd/blob/master/main/app/reactiveLayer/allostaticController/conf/default.ini)). 

## Still to be done: 
- command line argument to specify the size of the time window
- to select which drives to plots (will be useful when we'll have more drives)
- plotting events like "behavior X starts", "behavior X stops" (e.g transparent rectangles on the plot)
- replay the animation since the beginning
- closing ports correctly (e.g. using the `KeyboardInterrupt` python exception) 
- update [`README.txt`](https://github.com/robotology/wysiwyd/blob/master/main/src/modules/reactiveLayer/README.md) with the above kind of documentation
- ... *tell me what else is important (or rather, is urgent).*

I'll be hardly reachable this Thursday (wysi pi meeting). Maybe @sockchinglow can help a bit if needed (I've rapidly explained the code to her, it is also a bit commented).

Cheers

Tobias-Fischer, 2016-02-29:
Hi @clement-moulin-frier and @sockchinglow,
Did you guys further work on the plotting utility? The to-do list looks pretty good to me, and it would be great if we could have some of the features by the start of the integration week. It'll be a helpful debug tool I guess.

Best,
Tobi

clement-moulin-frier, 2016-03-04:
Hi @Tobias-Fischer ,

Yes we'll improve it before the integration meeting, in particular adding behavior start and end events.

Tobias-Fischer, 2016-03-16:
Hi @clement-moulin-frier,
We briefly talked about this yesterday. I guess we are very close to what we need. The only useful thing would be to have the rewind feature so we can see where things might have gone wrong.

Taking the high priority label off ..

Thanks, Tobi

Tobias-Fischer, 2016-03-17:
Hi @clement-moulin-frier,
The following would be nice:
1) Being able to start the script without having `allostaticController` running before. It should just wait and try to connect in a loop. If it looses connection because `allostaticController` was closed, it should go back to start and wait again for the connection
2) We are (sometimes) unable to close the script. Do you catch CTRL+C?

clement-moulin-frier, 2016-03-17:
I updated the module yesterday now using the RFModule class, seems to solve the exit issue. I'm on my way to Inserm, I'll push the changes from there.

clement-moulin-frier, 2016-03-17:
pushed

clement-moulin-frier, 2016-03-18:
The reconnexion issue is addressed by c3145f1806fecabfe7508d1666928eb932f4b74b

For closing the script, using the `RFModule` class helps a bit but when `CTRL-C` occurs, it blocks in [interruptModule()](https://github.com/robotology/wysiwyd/blob/master/main/src/modules/reactiveLayer/allostatic_plot.py#L136-L144). Strangely, this also happen when this method simply return true.

Tobias-Fischer, 2016-03-20:
Hi @clement-moulin-frier,
Could you please open another issue in `yarp` asking for advice how to deal with the blocking behavior of `interruptModule`?

Thanks,
Tobi

Tobias-Fischer, 2016-03-21:
@clement-moulin-frier: Also, can you please check whether the problem with 100% CPU usage was fixed with the commit having a delay after each reconnect try? If it is fixed, we should add this script to the yarpmanager script @pattacini

maxime-petit, 2016-03-23:
Hi!

@gregoire-pointeau  @AnneLaureM : can you check that you have all the dependencies (python, ...) to launch the gui with the Rococo machine?

gregoire-pointeau, 2016-03-23:
Yes, `allostatic_plot.py` launch normally on `rococo`.
all the displays are now on `rococo` and working fine. We will however reduce the size of the histogram if we want to be able to show the drives.

maxime-petit, 2016-03-23:
Nice!

@gregoire-pointeau  Can you run also the allostatic_plot.py for 5 min and check the CPU usage on Rococo? And check please that you are not experiencing disconnection from the yarpserver. 

./Max

clement-moulin-frier, 2016-03-23:
On my computer there is no 100% CPU usage, even before the change I made with the delay (it's rather around 6% in both conditions).
I'm testing with dummy drives and not on the full demo (no access to the icub nor a speech machine where I am). But even if it is the case only with the demo I don't think it is an issue of the `allostatic_plot` alone.

clement-moulin-frier, 2016-03-23:
@Tobias-Fischer : I've addressed the `interruptModule` issue in 026dfe20f4f73407f397702a633266eafe534507 and eba614d4137a3e2dd69c09e3db12127a2550e615.

The problem was that I was using the [`matplotlib.animation.FuncAnimation`](http://matplotlib.org/api/animation_api.html) which was blocking the execution. I don't use this class anymore but the [matplotlib interactive mode](http://matplotlib.org/api/pyplot_api.html#matplotlib.pyplot.ion) instead.

The module now closes without trouble on my computer. Let me know if it is fine on your side.

maxime-petit, 2016-03-23:
Actually, was it a CPU overload or a bandwidth overload? The main problem was that it was making the all computer disconnected from the yarp server

./Max

Tobias-Fischer, 2016-03-23:
Hi @clement-moulin-frier, thanks for getting the script to close properly!

@maxime-petit We had seen a CPU overload. I guess this happened when the script couldn't connect to any drives as no drives were running and it tried to reconnect with high frequency. This then also lead to disconnects with the yarp server.

Tobias-Fischer, 2016-03-27:
Closing here, as all issues are addressed as far as I can tell. Thanks @clement-moulin-frier!

Tobias-Fischer, 2016-03-31:
@maxime-petit and I just discovered that the problem that `allostatic_plot.py` kills the yarpserver still persists. It would be great if you look into this as fast as possible.

Additionally, no GUI is shown on our machine.

Best, Tobias

clement-moulin-frier, 2016-03-31:
Hi @Tobias-Fischer 

We've just tried and it works well. Are you sure that the problem is not the other way round: because your yarpserver is killed (for whatever reason), the `allostatic_plot` doesn't show anything?

Tobias-Fischer, 2016-03-31:
Hi @clement-moulin-frier,
The yarpserver is definitely killed by `allostatic_plot`. We've been running lots of experiments without `allostatic_plot` and all is running fine in this case. My guess is that something goes wrong when you kill other modules and `allostatic_plot` stays active in the meantime.
Also, `allostatic_plot` does not plot anything here right from the beginning. Checking `matplotlib` in a terminal however works fine.

clement-moulin-frier, 2016-03-31:
OK, please tried with the fix in 1fff4df76629b56049f2418488da2932c40efbd0

Explanation:

We observed that the `allostatic_plot` is slowing down when the `yarpserver` becomes more loaded (e.g. when launching `iol`). 
We also observed that when we don't try to automatically reconnect to disconnected ports, there is no slow down, even when no port is actually disconnected. This means that what is slowing down the module is the call to the `Yarp::is_connected` method, which can be due to an overload of the `yarpserver`.

To fix this we did two things:
- We try to reconnect to ports only every 10 seconds
- We add a delay of 0.1s after each call to `Yarp::connect`

A remark: I'm not convinced that it is a good practice to systematically try to reconnect to ports. This (1) complexify the code (2) overload the yarpserver and (3) can introduce bugs that will be hard to detect (e.g if we forgot to code the reconnexion to a freshly introduced port in a module).
So the question is (in particular to @pattacini ): is this practice standard in the yarp community and is it a good practice? In my opinion it leads to the same issues than when using the persistent ports (see #85 ), i.e. that we are loosing control on the connexions. Instead I think we should ensure that the modules have dependencies and should be launched in a specific order.

Best :)

Tobias-Fischer, 2016-03-31:
Hi @clement-moulin-frier,
Thanks for the fix! We'll try tomorrow.

Regarding the port connections: It is definitely not standard to automatically reconnect. Nor to use persistent connections. The standard is to check whether the dependencies are running, and if so, to do whatever needs to be done. If the dependencies are not running, you just wait. `yarpmanager` is used to connect / disconnect. So the order in which the modules are launched should not matter ..

pattacini, 2016-03-31:
I endorse @Tobias-Fischer's answer.

The policy is that a module should not perform connections with external ports autonomously, but rather adheres to the _observer_ principle, being as much agnostic as possible about who's going to talk to it or whom it is going to talk to. Therefore, connections are established outside the module and thus the module itself should deal with conditions where there are no data to process for example, meaning you have to skip any processing and/or wait for the connection being made.

That being said, there exist exceptions to this golden rule of course, but they should be handled carefully and case by case.

##### Question
Before the fix, did the module keep connecting/disconnecting the ports all the time? If so, that was really overloading the network. Also, bear in mind that, currently, if yarp is asked to make a connection between two ports that are already connected it will first disconnect them, causing a further burden (there's no easy solution to that). You could then first check if they are already connected and skip the connection.

pattacini, 2016-03-31:
Lately, we have also spotted this problem while disconnecting a port: https://github.com/robotology/yarp/issues/715.

So, be careful :smirk:

clement-moulin-frier, 2016-03-31:
The logic makes complete sense indeed but I'm afraid we are here in a special case. This module gets the list of drive and behavior names and then has to connect to the appropriate ports accordingly, meaning that the ports to connect are only known at run time. We could still set the connections in the `yarpmanager`, but this means that we'll have to modify the `xml` each time we want to add/remove a drive or a behavior, not sure we want this. 
This is also maybe a mistake in our conception of the `allostaticController` module, maybe we could use one single output port instead of one per drive. We'll definitely rethink the logic behind our modules after the review meeting, it should simplify everything. 

To reply to your question @pattacini : we didn't observe any intempestive connection/disconnection (none at all actually). For reconnecting to ports,  we use the method you mention: first check if they are already connected using `isConnected` and connect only if not. But what we observed is that those `isConnected` (a dozen of connections to check let's say), is slowing down the module.

Tobias-Fischer, 2016-03-31:
So where does the list of the drives and behavior names come from? I guess a config file? In this case, why don't you read the config file (of the other module)?

Tobias-Fischer, 2016-03-31:
Also, you could also just wait until the port where you read the behavior names from becomes available ..

pattacini, 2016-04-01:
The idea @Tobias-Fischer put forward is certainly doable and could give a solution to this problem with the least effort. Anyway, going for one single port that connects to all the downstream modules should be really preferable (maybe in long term?). Actually, the single-port approach is the most common.

Was there any particular reason why at design stage you chose the solution foreseeing multiple connections? With a very light protocol you can easily rely on one port indeed.

clement-moulin-frier, 2016-04-01:
@Tobias-Fischer the names come from specific rpc commands to the corresponding modules, but yes they could also come from the reading of the associated config file. But still the names of the ports contain the names of the drives/behaviors, so we now them only at run time. This is the problem and we'll adopt the single-port solution @pattacini proposes, which should solve the problem and allow to deal with connections only in the `xml`. However it will be complicated to do it before the review, rather just after with other refactoring of UPF's modules to start year 3 on more robust bases.

Tobias-Fischer, 2016-04-01:
Okay. Let's chat about this at the review meeting, and keep the code as is for now :)

Tobias-Fischer, 2016-04-01:
It seems like that nothing is shown is a `matplotlib` problem. Even very simple scripts don't show a window, it's just flickering up at the very end.
Not sure how to solve this.

Tobias-Fischer, 2016-04-01:
Solved with https://github.com/robotology/wysiwyd/commit/bb88e2991dc98335b1612f2026d5c6e96184d614

closed 2016-04-01 by Tobias-Fischer