robotology-legacy/wysiwyd #68: Try SAM with OpenCV3

Tobias-Fischer, 2016-01-21:
See whether SAM still behaves the way it should using OpenCV3 (see https://github.com/robotology/wysiwyd/pull/67).

Tobias-Fischer, 2016-02-14:
Hi @dcam0050, did you get a change to try this?

Thanks, Tobi

Tobias-Fischer, 2016-02-29:
Any news @dcam0050?

Tobias-Fischer, 2016-03-17:
@dcam0050: Can you please test this tomorrow so we can close this before the end of the integration meeting?

Tobias-Fischer, 2016-03-31:
I'll try it one last time @dcam0050.

Tobias-Fischer, 2016-04-01:
Closing as no one seems to care. Don't blame me if it's causing problems in the future ..

pattacini, 2016-04-02:
I've just notified @dcam0050 via email.

dcam0050, 2016-04-02:
Sorry @Tobias-Fischer  the last notifications you sent were sent to the now closed student email address I had. I sincerely apologise for this!

SAM with OpenCV3 will not be a problem for the integration meeting because we are currently not using vision driver which depends heavily on opencv2. I have noted this issue and when the time comes to resurrect that project will make sure everything works with opencv3.

So we can close this issue. Sorry again for the very late reply.

closed 2016-04-02 by dcam0050