robotology-legacy/wysiwyd #71: Resolve pitfall on opcclient::changeName vs. iol2opc::changeName

pattacini, 2016-01-26:
As @Tobias-Fischer correctly pointed out, we should resolve asap the potential pitfall we have on the name-change procedure, since we can do that in two different ways: through `opcclient` and through `iol2opc`.

pattacini, 2016-01-26:
At first sight, it seems that we always rely on `iol2opc::changeName()` for this service when we deal with objects and we resort to `opcclient::changeName()` only when the `iol2opc` sub-system is not available.

@Tobias-Fischer is this sufficient?

Tobias-Fischer, 2016-01-27:
Yes this is sufficient. I think we should remove the call to `opcclient::changeName()` in case that the `iol2opc` system is available, as the latter now removes stale objects automatically which was not the case before.
I'll do this in a minute.

Tobias-Fischer, 2016-01-27:
See https://github.com/robotology/wysiwyd/commit/1bde1583f4bc36da62fa84df712a526527fc141e

I think we can close here.

closed 2016-01-27 by pattacini