robotology-legacy/wysiwyd #72: Ears segfaulting and SpeechRecognizer blocking

jypuigbo, 2016-01-26:
I've been tracking the issue of ```ears``` unstability and I have maybe one first insight. 

```ears``` is currently trying to recognize something and it doesn't stop until it recognizes something This makes it a blocking module (so also slowing the flow of the ```proactiveTagging``` module). I added a couple of fixes that make it segfault less often, although don't solve this specific issue. 

I wanted to know if anyone here has tried or knows how the asynchronous recognition from the ```speechRecopgnizer``` works and if you think it might be a good alternative to solve the issue. 

This is more to open the discussion of what's the best way to proceed.

Tobias-Fischer, 2016-01-26:
See also https://github.com/robotology/wysiwyd/issues/50

Tobias-Fischer, 2016-01-27:
Did you push the fixes for `ears`?

jypuigbo, 2016-01-27:
I think I committed but not pushed. Anyway, before I do, yesterday a discussion begun with @clement-moulin-frier : are we sure that 
```
if (yarp::os::Network::connect('partNameA', portNameB')){ do_something();}
```
does the same as
```
if (yarp::os::Network::isConnected('PA','PB')
{
    do_something();
}else{
yarp::os::Network::connect('PA', PB');
do_something();
}
```
Because I was seeing ```subsystem_recog``` disconnecting and reconnecting to ```ears```, when ```proactiveTagging``` was loading a new grammar. 

When this is clear, I'll push or change the commit.

jypuigbo, 2016-01-27:
About the asynchronous part, I'll try to see if I can get how it works, as there's an rpc command to load/clear grammars asynchronously in ```speechRecognizer```.

Tobias-Fischer, 2016-01-27:
I am not sure how the yarp internals work in these regards. Do you know @pattacini? Can you give us some insight in what you are trying to achieve?

pattacini, 2016-01-27:
I think that a call to `yarp::os::Network::connect()` will first disconnect (if already connected) and then connect (again), whereas `yarp::os::Network::isConnected()` just queries the current status of the connection.

jypuigbo, 2016-01-27:
Cool. I made the changes to avoid this dis/re-connecting happen again. 

I leave the issue open to continue working on the asynchronous recognition part.

Tobias-Fischer, 2016-01-27:
See https://github.com/robotology/wysiwyd/commit/bb849b72f5d7cf1ac35f3623e090176eecbd7cf6

Tobias-Fischer, 2016-02-29:
Hi @jypuigbo,
Do you plan to continue working on the asynchronous recognition part? If not, can we close here?

Best, Tobi

Tobias-Fischer, 2016-03-07:
Closing. Feel free to reopen @jypuigbo

jypuigbo, 2016-03-15:
The problem seems to be tracked down thanks to @maxime-petit . 
I'm writing and issue to speech. The problem seems to be that, having two external grammars loaded, it keeps both active and doesn't send the output to both but to the last port. 

As soon as I get the post done I edit this one with the link.

jypuigbo, 2016-03-15:
After testing, seems that the issue is no longer there. Pull requested in [[SpeechRecognizer] Added option to interrup #6](https://github.com/robotology/speech/pull/6). Pending confirmation.

closed 2016-03-15 by Tobias-Fischer