robotology-legacy/wysiwyd #73: ProactiveTagging segfaults after successful recognition

jypuigbo, 2016-01-26:
I've updated all yarp and iCub today and it seems that the ```proactiveTagging``` module is no longer working in Barcelona. 

2 strange things happen: 
1) after recognizing the sentence, it prints it and has this shape: 
```
[DEBUG]Listen got reply ack 
[INFO] Reply from Speech Recog :  ACK this -1.0 is -1.0 a -1.0 pig -1.0 
```
2) After printing this, it consistently segfaults. 

I wanted to know if this is known or it might be related to the update.  
@Tobias-Fischer @gregoire-pointeau

clement-moulin-frier, 2016-01-26:
A possible cue about why we have those `-1.0` in the output of the speech recognizer (we don't remember having seen them before):

It seems to come from [those lines](https://github.com/robotology/speech/blob/master/speechRecognizer/src/SpeechRecognizerModule.cpp#L741-L751), meaning that `USE_LEGACY` is `true`. I don't what this legacy is about, but in [an old wysiwyd xml](https://github.com/robotology/wysiwyd/blob/26c95e03fbfa82ddc3b41607980f0bb1b42ac18e/main/app/demos/scripts/wysiwyd.xml#L64-L68) the speech recognizer is launched with the `--noLegacy` option.

Maybe it can help ... or not.

Tobias-Fischer, 2016-01-26:
Maybe @maxime-petit can help out regarding the speech recognizer. Didn't you have problems regarding this recently? How did you solve the other issue?

Tobias-Fischer, 2016-01-26:
By the way, these kind of problems are exactly the ones I was referring to the other day on slack ...

gregoire-pointeau, 2016-01-27:
I confirm this comes from the `speechRecognizer` and not from `proactiveTagging`.
I mixed up between what is legacy and what is not, but for sure Clément you'll have to play with it.

maxime-petit, 2016-01-27:
Yop!

speechRecognizer should always be running with --noLegacy as option. I thought I updated the .ini file but it seems not :)

I opened an issue to have this option by default

./Max

pattacini, 2016-01-27:
Hold on @maxime-petit ;)
Default options of `speechRecognizer` won't change until I'll be alive :)
As far as I understood its behavior can be already tuned via command line, so in `wysiwyd` we stick to the present configuration.

Tobias-Fischer, 2016-01-27:
This issue might be related, where @jypuigbo reported problems but then never replied whether it was resolved .. https://github.com/robotology/wysiwyd/issues/25

maxime-petit, 2016-01-27:
@ugo : no problem :)

pattacini, 2016-01-27:
:+1: :smirk:

jypuigbo, 2016-01-27:
The --noLegacy effectively solved the problem. Thanks!

closed 2016-01-27 by jypuigbo