robotology-legacy/wysiwyd #74: caffeCoder starts with a segFault

gregoire-pointeau, 2016-01-27:
Hi. Everything is in the title, here is was I get when launching `caffeCoder` on the `icub-cuda`:

> icub@icub-cuda:/usr/local/src/robot$ caffeCoder
||| clearing context
||| adding context [himrep]
||| configuring
||| no policy found
||| default config file specified as caffeCoder.ini
||| checking [/usr/local/src/robot/caffeCoder.ini] (pwd)
||| checking [/home/icub/.config/yarp/robots/iCubLyon01] (robot YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/robots/iCubLyon01] (robot YARP_DATA_HOME)
||| found /home/icub/.local/share/yarp/robots/iCubLyon01
||| checking [/etc/yarp/robots/iCubLyon01] (robot YARP_CONFIG_DIRS)
||| checking [/usr/local/src/robot/yarp/build-x86_64/share/yarp/robots/iCubLyon01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-x86_64/share/iCub/robots/iCubLyon01] (robot YARP_DATA_DIRS)
||| found /usr/local/src/robot/icub-main/build-x86_64/share/iCub/robots/iCubLyon01
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubLyon01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/wysiwyd/main/build/share/wysiwyd/robots/iCubLyon01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/yarp/build-x86_64/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-x86_64/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/wysiwyd/main/build/share/wysiwyd/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/icub/.local/share/yarp/robots/iCubLyon01/caffeCoder.ini] (robot)
||| checking [/usr/local/src/robot/icub-main/build-x86_64/share/iCub/robots/iCubLyon01/caffeCoder.ini] (robot)
||| checking [/home/icub/.config/yarp/contexts/himrep] (context YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/contexts/himrep] (context YARP_DATA_HOME)
||| found /home/icub/.local/share/yarp/contexts/himrep
||| checking [/etc/yarp/contexts/himrep] (context YARP_CONFIG_DIRS)
||| checking [/usr/local/src/robot/yarp/build-x86_64/share/yarp/contexts/himrep] (context YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-x86_64/share/iCub/contexts/himrep] (context YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/contexts/himrep] (context YARP_DATA_DIRS)
||| found /usr/local/src/robot/iCubContrib/share/ICUBcontrib/contexts/himrep
||| checking [/usr/local/src/robot/wysiwyd/main/build/share/wysiwyd/contexts/himrep] (context YARP_DATA_DIRS)
||| checking [/home/icub/.local/share/yarp/contexts/himrep/caffeCoder.ini] (context)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/contexts/himrep/caffeCoder.ini] (context)
||| found /usr/local/src/robot/iCubContrib/share/ICUBcontrib/contexts/himrep/caffeCoder.ini
Segmentation fault

`SFM` is running fine, and `icub-cuda` should be up to date (maybe not caffe?)

@pattacini @Tobias-Fischer

Tobias-Fischer, 2016-01-27:
Before doing any debugging, can you please update `himrep`? Also, did you follow the instructions in https://github.com/robotology/himrep/blob/master/README_Caffe.md (everything which is written under `caffeCoder Setup`?

GiuliaP, 2016-01-27:
Hi, may I ask you to perform some checks in order to try to understand the problem?

1) Does the error repeat the same even after cleaning all ports in the network?
2) Although it does not seem to be a problem of Caffe, which Caffe commit are you using? You can get it by tiping `git log -1` from Caffe's repository directory
3) Could you insert some printouts in the `.configure` method of the class `CaffeCoderModule` (lines `349-364` in the file `himrep/modules/caffeCoder/src/main.cpp`), rebuild and run? I need to understand at which instruction the error arises. For instance, before or after line 356 `caffePort = new CaffeCoderPort(rf);`. For sure, no ports are opened thus I suspect the problem is before line `358`. If you cannot do this, I can try to reproduce the error (but I guess it will take a little more time).
4) If you are in the need to run the module right now, you can try to (i) update Caffe bu pulling latest commit (but I strongly suggest you to keep a backup version of the Caffe version which you are using now) and (ii) run this PR (https://github.com/GiuliaP/himrep) instead of the master branch of `himrep`

GiuliaP, 2016-01-27:
Thanks @Tobias-Fischer, indeed your checks are to be done first :)

gregoire-pointeau, 2016-01-27:
Ok, My bad, it was coming from a mess between the different blades of the cluster and the cuda.
Thanks anyway !

closed 2016-01-27 by gregoire-pointeau