robotology-legacy/wysiwyd #75: Removing Choregraphie and Posture

gregoire-pointeau, 2016-01-28:
Hello.
Does anyone in @robotology/wysiwyd-developers use the `choregraphie.ini` and `posture.ini` from `iCubClient`?

I now that it was something created by Stéphane, but I'm pretty sure that no one uses it anymore. What about removing it and simplify all the config files?

Tobias-Fischer, 2016-01-28:
Fine with me.

gregoire-pointeau, 2016-01-28:
The idea would be also to put the list of `subsystem` which is until now, in a `client.ini` file, directly in the `module.ini` file.

Tobias-Fischer, 2016-01-28:
Then, also all of this should be removed: https://github.com/robotology/wysiwyd/search?utf8=%E2%9C%93&q=posture.ini&type=Code

Tobias-Fischer, 2016-01-28:
Regarding `client.ini` and `module.ini`, I don't know how the internals work, and propose to keep everything as is to not introduce more issues than we already have. However, if someone knows and is willing to change all configuration files of all modules, go ahead ;).

gregoire-pointeau, 2016-01-28:
I'm not certain that the `postures` present in the `client` and in the `Babbler` are related. I'mon the contrary pretty sure they are not.

gregoire-pointeau, 2016-01-28:
I just want to be sure that no one "really" uses the `client.ini` file for anything else that the list of `subsystems` and that no one uses the `posture` and `choregraphie` from the client.

Tobias-Fischer, 2016-01-28:
All I wanted to say with the link is that if we decide to remove the config files, we should also remove the related code.
Maybe @stephane-lallee can provide some insight to these three ini files?

gregoire-pointeau, 2016-01-28:
I am in `iCubClient.h` and `iCubClient.cpp`  trying to figuring it out.

gregoire-pointeau, 2016-01-28:
I've checked, the only files `posture.ini` and `choregraphie.ini` found are in `iCubClient` and the only use if found in `postureBabller`.
I'm almost certain that this module is no more used in WYSIWYD and we could remove it.

clement-moulin-frier, 2016-01-29:
We'll probably use such postures and choregraphies for the reactive layer but not those defined in the files you mention, so why not removing them if you want. However we sometime use the `postureRecorder` so we should keep the related code (as well as the code to replay them). Cc @jypuigbo

gregoire-pointeau, 2016-01-29:
I just wait for @jypuigbo and I remove `posture.ini` and `choregraphie.ini`.

stephane-lallee, 2016-01-29:
You'll probably need to change the iCubClient constructor as it is expecting those. Have fun ! 😉 

-----Message d'origine-----

Tobias-Fischer, 2016-01-29:
I agree with @stephane-lallee that it is not enough to just remove these two files. I guess you would also need to remove the related code in `iCubClient` and wherever else it is used (god knows ..).

stephane-lallee, 2016-01-29:
Don't remove all the related code. Just make the loading of those file optional when initialising the client.
If the fields for the files a filled then load them, if not then just skip the part where the data structures are created. They are just populating two lists if I remember correctly.

-----Message d'origine-----

gregoire-pointeau, 2016-02-05:
Done, loading of choregraphies and postures are now disabled by default.
see : https://github.com/robotology/wysiwyd/commit/63d2017411f55817c511f01f9b44fc1416f00b6a

Tobias-Fischer, 2016-02-05:
Can you please change postureBabbler to load these files? I know we don't use it right now, but let's not knowingly break any more modules.

gregoire-pointeau, 2016-02-05:
Done, see: https://github.com/robotology/wysiwyd/commit/87e99aa3b23999cafe44c920ced2c3fa2655f7f4
When you declare your `iCubClient`, there are now 2 boolean `false` by default, just set them `true` to use it.

gregoire-pointeau, 2016-02-05:
please @Tobias-Fischer , give it a try and tell me if it load everything correctly.

closed 2016-02-05 by gregoire-pointeau

Tobias-Fischer, 2016-02-05:
Didn't test, but it looks okay. Thanks.