robotology-legacy/wysiwyd #76: bodySchema launching problem with opencv

gregoire-pointeau, 2016-02-10:
Hi.
I updated opencv on the `icub-cuda` machine, but now, I have the following problem when I launch `bodySchema`:

`icub@icub-cuda:~$ bodySchema`
`bodySchema: error while loading shared libraries: libopencv_videostab.so.2.4: cannot open shared object file: No such file or directory`

But `iol2opc``works fine. DO you know why it could have such a problem of `opencv` lib on the `icub-cuda`?

Tobias-Fischer, 2016-02-10:
Hi,
why do you want to run `bodySchema`? This is an old module and was split: https://github.com/robotology/wysiwyd/tree/master/main/src/modules/bodySchema
Most likely, you want to run `babbling` (see `proactiveTagging.xml`).
The error occurs because the `bodySchema` executable which is still around is linked to an old version of OpenCV.

Best, Tobi

closed 2016-02-10 by Tobias-Fischer

gregoire-pointeau, 2016-02-10:
Okay nice.
I was trying all modules and I had a doubt about this one.
Ok, problem solved then.