robotology-legacy/wysiwyd #77: Table with comparisons to other projects

Tobias-Fischer, 2016-02-16:
We need to put the outcomes of https://github.com/robotology/wysiwyd/issues/55 https://github.com/robotology/wysiwyd/issues/59 https://github.com/robotology/wysiwyd/issues/57 https://github.com/robotology/wysiwyd/issues/56 https://github.com/robotology/wysiwyd/issues/54 https://github.com/robotology/wysiwyd/issues/53 together in tables showing the differences to WYSIWYD on one glance. We need to decide on the format of the table, and need to transfer the loosely written reports in these tables.

Best, Tobi

sockchinglow, 2016-03-15:
I'm putting together the reviews into one document; I'll post follow-ups henceforth on the dedicated slack channel.

Tobias-Fischer, 2016-03-15:
That's great news, thanks!

sockchinglow, 2016-03-15:
The workload has been assigned/completed in the following manner.
[_CHRIS_](http://www.chrisfp7.eu/): Ugo
[_RoboHow_](https://robohow.eu/): Hyung Jin
_Cangelosi_: Tobias
[_POETICON(++)_](http://poeticon.csri.gr/): Maxime
[_ALEAR_](https://ikw.uni-osnabrueck.de/~neurokybernetik/projects/alear.html): Maxime
[_iTalk_](http://www.italkproject.org/): Jordi

Also included are some projects listed in the DoW.
[EFAA](http://efaa.upf.edu/project)
[JAMES](http://homepages.inf.ed.ac.uk/rpetrick/projects/james/)
[ROSSI](http://cadair.aber.ac.uk/dspace/handle/2160/553)
[CogX](http://cordis.europa.eu/project/rcn/86240_en.html)
[HUMANOBS](http://www.humanobs.org/)

I have posted the link to the table on slack, please refer to it. This is especially for the projects that you personally reviewed (@hjchang, @Tobias-Fischer, @maxime-petit,  @jypuigbo ), because some elements may be missing from the overview given and I might have misunderstood some details. Also, the other projects that were retrieved from the DoW needs a look-through.

@pattacini Still awaiting your review of CHRIS.

Lastly, STRAND was mentioned recently in the slack channel. Shall we include that as well?

Tobias-Fischer, 2016-03-16:
Hi @sockchinglow,
Thanks, this is really helpful. Great.

Regarding STRAND, there is no need to include it.

@hjchang @maxime-petit @jypuigbo @pattacini: Please drop a quick message here after checking the table so we can close this issue.

Thanks,
Tobi

pattacini, 2016-03-16:
Fine with me.

hjchang, 2016-03-16:
All right!

Tobias-Fischer, 2016-03-16:
Fine with me.

Tobias-Fischer, 2016-03-17:
@maxime-petit @jypuigbo @pattacini: Can we close here?

pattacini, 2016-03-17:
Yes, I think

sockchinglow, 2016-03-18:
On @clement-moulin-frier's suggestion, I have added another two columns: Develops an autobiographical narrative; Learns to distinguish between the self and others.

Could contributors fill in those cells as well?

Also, it may be a good idea for us to reach a consensus regarding WYSIWYD's markups in particular.

sockchinglow, 2016-04-01:
Everyone except @pattacini has yet to complete the extra columns. Would be good if it can be completed soon.
@Tobias-Fischer @jypuigbo @maxime-petit @hjchang

Tobias-Fischer, 2016-04-02:
Hi, I've added some more stuff for Cangelosi + Poeticon++. However, I am not sure about some of the columns and left them empty rather than putting in false information. Would be good if someone else can have a look (maybe @maxime-petit?)

Tobias-Fischer, 2016-04-06:
Hi @maxime-petit,
I saw you working on this yesterday. Did you finish it?
@jypuigbo @hjchang: Did you fill the table?

Best, Tobi

jypuigbo, 2016-04-06:
Yes I did, if nothing changed :)

Tobias-Fischer, 2016-04-07:
This should be done now. @sockchinglow: Do you want to send an email around to the WYSIWYD mailing list containing a link to the table and check that it's with with the PI's?

Tobias-Fischer, 2016-04-15:
@sockchinglow: Please send this around to the mailing list ASAP!

clement-moulin-frier, 2016-04-15:
I've just done it through the maillist `wysywid-project`

closed 2016-04-15 by Tobias-Fischer