robotology-legacy/wysiwyd #78: babbling missing argument

gregoire-pointeau, 2016-02-16:
Hi.
When I launch the proactiveTagging demo with unknown body part, I have the following with `babbling`:
`command received: babbling joint 9`
`[Error]Invalid babbling part: specify LEFT or RIGHT after joint number`

Where this specification should be done? in `proactiveTagging.ini` ?
Thanks @Tobias-Fischer @martz21

Tobias-Fischer, 2016-02-16:
This is because of the recent changes by @martz21. Can you please have a look?

Best, Tobi

maxime-petit, 2016-02-16:
Yop!

I changed the babbling subsystem. I add a default value for the side of the arm ("left") and did not change the code in proactive. 
Another way should have been to modify the code in proactive and NOT put any default value for the subsystem. Do you think this way is better?

Let me know and I ll change it if needed

./Max

gregoire-pointeau, 2016-02-18:
Hi Maxime, indeed now the robot is moving !
I think when you create the subsystem you should be able to select the hand (with a default value of course) because here, we would need to use the right hand (we have a problem with the skin of some some finger tips).

So: a default value, that could be change in the conf file of the module (and not the subsystem). Is it doable?

maxime-petit, 2016-02-18:
Yop

Should be done!

Check please

./Max

Tobias-Fischer, 2016-02-19:
Can we close this @maxime-petit?

gregoire-pointeau, 2016-02-19:
On my side, it's fine !

closed 2016-02-19 by gregoire-pointeau