robotology-legacy/wysiwyd #80: Getting more done in GitHub with ZenHub

Tobias-Fischer, 2016-02-29:
Hola! @Tobias-Fischer has created a [ZenHub](http://www.zenhub.io) account for the **robotology** organization. ZenHub is the leading team collaboration and project management solution built for GitHub.

----

#### How do I use ZenHub?

To get set up with ZenHub, all you have to do is **[download the browser extension](https://www.zenhub.io?utm_source=ZHOnboarding)** and log in with your GitHub account. Once you do, you’ll get access to ZenHub’s complete feature-set immediately.

#### What can ZenHub do?

ZenHub adds a series of enhancements directly inside the GitHub UI:

- Real-time, customizable task boards for GitHub issues;
- Burndown charts, estimates, and velocity tracking based on GitHub Milestones;
- Personal to-do lists and task prioritization;
- “+1” button for GitHub issues and comments;
- Drag-and-drop file sharing;
- Time-saving shortcuts like a quick repo switcher.

### [Add ZenHub to GitHub](https://www.zenhub.io?utm_source=ZHOnboarding)

_Still curious? See [more ZenHub features](https://www.zenhub.io/features?utm_source=ZHOnboarding) or read [user reviews](https://chrome.google.com/webstore/detail/zenhub-for-github/ogcgkffhplmphkaahpmffcafajaocjbd/reviews). This issue was written by your friendly ZenHub bot, posted by request from @Tobias-Fischer._

[image]

Tobias-Fischer, 2016-02-29:
Hi guys, this might be a nice thing to have during the integration meeting. It's just a browser extension, but seems to be quite powerful. Thanks @pattacini for the tip.

pattacini, 2016-02-29:
`Chrome` and `FireFox` are currently supported.

clement-moulin-frier, 2016-03-15:
Hi guys,

I've started to use the Zenhub board to organize the issues we have to handle during the integration. Please install the extension (see @Tobias-Fischer message above) and check the freshly added `Boards` tab on the wysiwyd github.

Clément

Tobias-Fischer, 2016-03-16:
Hi @robotology/wysiwyd-developers,
Last reminder to please install Zenhub.

Closing here.

closed 2016-03-16 by Tobias-Fischer