robotology-legacy/wysiwyd #81: Handle more actions

clement-moulin-frier, 2016-03-15:
For the time being we just deal with `pointing`. Try to add a few other actions, e.g. `looking`, `pushing` or `grasping` depending on what ARE is able to execute safetly.

See also #45

sockchinglow, 2016-03-16:
Creating a behaviour called glance for looking at something/someone for ~0.5seconds.

jypuigbo, 2016-03-16:
We are solving this issue temporarily, as we have already some code regarding this, that we had no time to properly debug before coming to the integration, which was adding extra functionality. 

Adding as features: looking at X and pushing X

Tobias-Fischer, 2016-03-16:
@jypuigbo @sockchinglow:
Can you please open a pull request so we can check this on the robot?

Best,
Tobi

sockchinglow, 2016-03-16:
Handled in #84

Tobias-Fischer, 2016-03-17:
Let's test this tomorrow morning, so we can close the issue.

jypuigbo, 2016-03-29:
Ears now can listen to point, push and look_at, triggering the corresponding actions. Tested and debugged at UPF. Let me know of any additional fix needed.

Tobias-Fischer, 2016-03-29:
Looking good. Thanks :)

closed 2016-03-29 by Tobias-Fischer