robotology-legacy/wysiwyd #85: Using persistent connections or not

clement-moulin-frier, 2016-03-17:
Hi,

@pattacini pointed me to the possibility of [using persistent connections in YARP](http://www.yarp.it/persistent_connections.html), that allows an automatic reconnexion whenever one port or the other is closed then reopened. It looks like an easy solution to avoid some issues we have about module dependencies.

However this solution could have side effects so we should think about where those connections can help us and where they should be avoid.

As an example, I tried to change all the connections made by `allostaticController` to `persistent` and some of them were not working correctly anymore (no idea about the reason). I then did it in only one of them and I obtained the expected behavior.

Tobias-Fischer, 2016-03-17:
Hi @clement-moulin-frier,
As @pattacini said these connections might cause troubles. And you already found that `allostaticController` does not work properly with them, let alone all the other modules which run in our demo.
Thus, I think we should not make any use of them.

Best, Tobi

clement-moulin-frier, 2016-03-17:
Not sure he said they can cause trouble, rather that we are loosing control by using them.

pattacini, 2016-03-19:
I also feel to discourage the use of persistent connections for the time being. As we discussed, there is a specific procedure to disconnect a persistent connection and if we do not take it into account there's the serious risk of loosing control of the architecture.

Tobias-Fischer, 2016-03-20:
I agree with @pattacini. Let's not use these connections, and rely on the connection list in `yarpmanager` instead. In the long term, we should change our modules to not automatically connect to others, and instead wait for the connection to be made outside.
Unfortunately, I don't see this happening anytime soon :(.

Closing here.

closed 2016-03-20 by Tobias-Fischer