robotology-legacy/wysiwyd #86: Sleep drives in proactive tagging

Tobias-Fischer, 2016-03-17:
Hi all,
Thanks to https://github.com/robotology/wysiwyd/commit/a41e73ce066826ecd865710f81c362e6ed6e299e we can sleep all drives. I think we should make use of this in proactive tagging. Whenever we execute `exploreUnknownEntity`, `searchingEntity` etc. we should pause the drives, and let them continue afterwards.

Best, Tobi

jypuigbo, 2016-03-18:
As I already mentioned, It doesn't fit very well with the idea behind the `allostaticController`, BUT for the demo I find is a good idea. 

Issue is then: 
The command `sleep all <time>` needs to be sent to the port `/homeostasis/rpc`. This will stop them from moving for `time` seconds. Still, if they are out of the comfort zone they'll probably be triggered. 
Should the command be sent through `allostaticController`, when the behavior is **triggered** or by `proactiveTagging` / `behaviorManager` / any other behavior generating module when the behavior **finishes**?

The command is that sending the rpc comand to that port. If you want it in `allostaticController` let us know, and we add it in the behavior. Otherwise, ask any question you might have.

clement-moulin-frier, 2016-03-18:
I've tried another solution in e6becb9c872fe5e4f4dade16f19a6268bd344d06 where all drives are frozen whenever a behavior starts, and all are unfrozen when it stops. Works well by testing on dummy drives and behaviors, let's try it on the demo (allostaticController config file already updated accordingly in the mentioned commit)

Tobias-Fischer, 2016-03-20:
Hi @clement-moulin-frier,
Thanks for the new freezing behavior. Does that mean we don't have to change `proactiveTagging` and other modules at all? That would be great.
However, it seems that https://github.com/robotology/wysiwyd/commit/e6becb9c872fe5e4f4dade16f19a6268bd344d06 only updates the config file. Where is the commit for the code implementing `freeze` and `unfreeze`?

Best, Tobi

clement-moulin-frier, 2016-03-23:
Hi @Tobias-Fischer 

Nothing has to be changed in other modules, the changes appear in the `allostaticController` config files, where all drives are frozen whenever a behavior starts and unfrozen when it ends, see e6becb9c872fe5e4f4dade16f19a6268bd344d06

The code implementing `freeze` and `unfreeze` is in this commit: 1d5144c2274facb077b38fd462fbcacf8168fcbe

I tested it with dummy drives and behaviors and it works well, but I'm not sure it has been tested on the demo. Do you know?

maxime-petit, 2016-03-23:
Yop!

Just to jump on @jypuigbo about the fact that freeze/unfreeze is not fitting with 'allostaticController' idea. If reviewers are asking this question, we can argue that for now, we have not implemented an interrupt mechanism for existing behavior, thus we need to freeze drives in order to NOT be in a situation when the robot has to trigger another behavior with highest priority during the execution of a lowest one. 

Does it make sense?

./Max

jypuigbo, 2016-03-23:
@maxime-petit With @clement-moulin-frier we thought of describing it as the fact that the execution of a drive inhibits the other drives, so they don't decay until the behavior is finished again. 
This addresses the fact that for the demo we need the behaviors not overlapping, and strengthens the role of the allostaticController in regulating the dynamics of the drives, as we regulate this directly through allostatic controller. 

Also what you say makes is a good point to justify it, and I take good note for the day of  the review :)
Thanks.

Tobias-Fischer, 2016-03-27:
Hi @jypuigbo and @clement-moulin-frier,
Did you get to test this in the proactive demo?

Best, Tobi

jypuigbo, 2016-03-28:
Not yet. We update on this tomorrow.

jypuigbo, 2016-03-29:
Tested and working within the PT demo. The changes were already pushed during the IM in Lyon. Just make sure you have in your config files ( `allostaticController` -> `default.ini`) `(freeze all)` and `(unfreeze all)` before and after triggering the behaviors, as in the apps folder.

Tobias-Fischer, 2016-03-29:
Cool. Closing here. Thanks!

closed 2016-03-29 by Tobias-Fischer