robotology-legacy/wysiwyd #89: Important: Integration meeting summary for PI's / next meeting

Tobias-Fischer, 2016-03-18:
Hi all,
Unfortunately we didn't have time to summarize our accomplishments of the integration. 

1. I suggest one of us should write an email around to summarize the meeting and inform the PI's. I'm happy to do that over the weekend, unless someone else wants to do that.
2. We need to decide when and for how long we should meet to finalize the demo.
3. I'd like to ask everyone to progress on the demo back in the individual labs. We're running out of time, and I'm afraid even at the next meeting we might find more nasty bugs which prohibit progress.
4. I think it's a good idea to create a new branch for development while using `master` purely for bug fixes. New features are definitely needed, however typically add new bugs. As we don't have much time left before the demo, new bugs however make our life too hard. What do you think @pattacini ?

Best, Tobi
PS: good seeing you all!

Tobias-Fischer, 2016-03-18:
/cc @robotology/wysiwyd-developers

pattacini, 2016-03-19:
Hi @Tobias-Fischer,
Thanks for the wrap up: we should really address these points asap, indeed.

Coming to them:

1. Agreed. I think we should also report _separately_ on the status of **Action Recognition**. In this respect, **PFD** already asked @dcam0050 to do that, and I think it'd be beneficial also to get a resume from @MagnusJohnsson and/or @zahrag as well.
2. I'd propose to meet up again on the **13th/14th of April**, conveniently in advance wrt the review.
3. Yes!
4. Definitely. We can do even better by creating the **`year-2-demo`** branch from master, which will contain the software we are going to show. We should put in this branch **bug fixes** and **new features** we are going to produce only within the next week. Then, the development **must be interrupted**. We could do _one exception_ for UPF, given that they're not available to code next week.

jypuigbo, 2016-03-19:
Hi, 
Thanks @Tobias-Fischer for the wrap up. 

On topic 4: 
Just to be sure on the procedure: 
- Regarding the issues: which priority of features need to be added: highpriority only? all? As much as possible?
- Are or will be all the features needed listed as issues, here in github? The same for bugs, etc. 
I think having everything here will make sure no one forgets anything, even if its for a quick fix, specially if its another lab the one who has to solve it. 

Thanks and my apologies for such obvious questions.

pattacini, 2016-03-19:
I think @jypuigbo's suggestion makes lots of sense. We should then detail in GitHub the **few  new features** we really want to have in the demo and that will be developed next week - I hope they'll be just a few - so as **all the bugs** we need to fix.

To summarize:

- The **`year-2-demo`** branch will contain **all the bugs** and the **few new features** we aim to have for the second year demo. Please open an issue for each bug/feature, label it as you like, but remember to put it into the correct **milestone**. Therefore, we should remove from the milestone those issues we are no longer interested in for this deadline.
- All the rest should go into `master` branch.

##### P.S.
In a usual git workflow `master` is used as _stable_ branch; however, I reckon that it would be safer to start with a different branch, at least for this time.

pattacini, 2016-03-22:
I think everything has been address: closing.

closed 2016-03-22 by pattacini