robotology-legacy/wysiwyd #90: Being able to manually control the drives

jypuigbo, 2016-03-21:
Hi all, 

Has been raised the topic to manually control the robot's drives. I think we have most of the tools already implemented, we just need to define exactly what we want/need and prepare an interface. 

My suggestion would be: 
- stop command in ears puts the drives of the robot on constant freezing mode, waiting for another human command to unfreeze. 
- Voice or RPC commands are set such that: we change the decays of the drives to meet specific values OR we set a drive under or over threshold so that the behaviors are triggered by `allostaticController` when requested. 
- (I don't know much about GUIs)

Let's discuss this and constrain the task as much as possible so the objective is clear and we can have it on time. 

(One drawback I foresee is that all tagging options are within the same drive, so we can't choose whether the robot will ask for partner name, objects or bodyparts in advance)

maxime-petit, 2016-03-21:
Hi,

* A pause mode can be handy if we need to explain/comment the demo :+1: 
* The simplest way I guess would be  to change the drive value directly (and not the decay) to be just above the threshold and indeed trigger it a few seconds after (so your second proposition for your second item :))
* RPC or GUI: RPC should be very quick to implement, and probably more useful in a debug/coding mode. However for the demo, I think we need to avoid "typing" command. I don't know much about GUI neither but @clement seems to have a more precise idea. 

For the drawback, indeed tagging can actually trigger several different tagging behavior depending on the kind of entity to tag. The choice of this entity is random right? However in practice, we can restraint the total number of entities to 1 agent, 1 body part and 2 or 3 objects. With body part, we have 2 tagging possibilities (names and tactile) but the others can only tag the names. So in total we can cover everything by running it 5/6 times which should be fine, especially when we know that object can be tagged only if present, so by removing them we force the robot to tag agent/bodypart, and this trick can be used in the "manual" triggering of specific drives.

./Max

jypuigbo, 2016-03-21:
@maxime-petit 

> The simplest way I guess would be to change the drive value directly (and not the decay) to be just above the threshold 
This is roughly what I meant. So we agree on that :)

clement-moulin-frier, 2016-03-21:
Hi guys,

I think the easiest way would be to directly acts on behavior, not drives. In particular because it allows to have a much better control on what we really want to launch, e.g. tagging a particular bodypart (instead of manually increasing the `tagging` drive value, which doesn't ensure what will be actually tagged). 
So I rather suggest that we (UPF) code a set of very constrained behaviors (e.g. "tag a bodypart"), that can be launched by RPC commands. I can then code a GUI in Python to send this RPC commands by pushing a button. Does it make sense?

@max: I agree a `pause` mode is needed. Since we coded a way to freeze all drives during the IM, this is quite straightforward to do.

Anyway, what we need is a list of behaviors we want to manually launch. Here is a first tentative, please update it:

- proactively tag an unknown object
- reply to a human order: involving a known objet vs an unknown object 
- proactively tag a body part
- Express knowledge about a body part
- Launch the narrative (possibly a limited version of it, should not last more than 1mn I would say)
- ...

maxime-petit, 2016-03-21:
Yop!

proactively tag a body part can trigger 2 different tagging : 
- the name if the name is *unknown*
- the tactile info if the name is known and `tactile_info == -1`. 
But the rpc command is exactly the same in both case `exploreUnknownEntity bodypart name_of_the_bp`. 

What do you mean as behavior for "express knowledge about a body part"?

./Max

clement-moulin-frier, 2016-03-21:
Regarding the different bodypart tagging procedures: then we should start the demo with a completely unknown body part (both name and tactile info unknown) and manually launch `exploreUnknownEntity bodypart name_of_the_bp` twice. 

Regarding "express knowledge about a body part", I was thinking about the behavior where the robot says something like "Do you know this is my index finger", together with moving it. (I remember having seen this but I'm maybe wrong, let me know in that case).

More generally, how about this structure for the review demo:

- First, we demonstrate each behavior manually using the aforementioned GUI:
    - We start with an OPC filled with unknown entities and manually launch: proactive tagging of an object, then of a body part (in two steps for the body part, first name then tactile), then knowledge expression ("do you know this is X"), then the narrative. Doing this manually with total control on what will happen requires that we know the `unknown_x` label of each entity in the OPC in advance, is it feasible?.
- Second, we explain that those behaviors can be triggered autonomously through allostatic control, but that in that case we don't have control on what will happen. Here we run the full demo, without wondering that much about the behavior sequence chosen by the robot since all behaviors will have been demonstrated in the previous point.

Tobias-Fischer, 2016-03-21:
Hi all,
Just quickly: I agree that this feature will be useful. I don't have time to comment on details.
One thing though: don't forget about the kinematic structure learning. Maybe @maxime-petit and @hjchang  can provide some details.

Best, Tobi

hjchang, 2016-03-21:
Hi!
I agree that manual triggering and pausing will be helpful. Also regarding the kinematic structure learning demo, I need the pasar to be paused for a while as it interrupts data retrieval from ABM. @clement-moulin-frier It will be nice to show the kinematic structure learning module running for a simple object structure as well during the first stage by launching the module from GUI. But I think it doesn't need to be a part of full demo scenario because it doesn't fit into the current scenario. It might be better to be another demo combining motor babbling, ABM, kinematic structure learning, and pointing a corresponding human body. 

Best, Hyung Jin

jypuigbo, 2016-03-21:
Doesn't it look nicer to first show the robot behaving by itself, also with the human giving him orders and after a while, the human stops it and we show the rest of the behaviors manually? Otherwise it looks like: "we have several independent modules and we will show you only some really integrated". Whereas the former is more like "Look at what we have done, but it will take long, and we have also the option to manually manipulate the robot, and we'll show additional nice stuff"

Besides, isn't it nicer if we say it as: "when we give him an order (spoken or GUI based) we are manipulating the drives in the allostatic controller, that is the same job future contextual layer will do". 

@clement-moulin-frier But if you want to call the behaviors directly, there's no need to create those behaviors, maybe it's better to directly send the RPC command to `proactiveTagging`, as the real behaviors are there (and `behaviorManager` is just a wrapper)

maxime-petit, 2016-03-23:
I agree with @jypuigbo with the order of the demo. Also in this case, the "manual" triggering will be only for unseen behavior and not all of them.

The question of calling directly the behavior or changing the drive to call immediatly the behavior should be "invisible" for external reviewer. Triggering the behavior automatically is more secure because as you said you bypass the drive, but changing the drive allow a nicer description I guess.

@clement-moulin-frier @jypuigbo @sockchinglow Maybe we can begin by the rpc command directly to the behavior and check the drive change after? In that case we will have a backup if the drives modifications imply unespected bugs?

./Max

jypuigbo, 2016-03-23:
@maxime-petit 
With @clement-moulin-frier we have been discussing how to address it and is as you say: rpc directly to the behaviors/specific functions to show, and I'll make that `allostaticController` also changes accordingly, but not triggering the behaviors from there. 

We get the visualization nice and the behavior expected, regardless of the actual commands behind.

clement-moulin-frier, 2016-03-23:
So we should now make a list of the individual behaviors we want to show, together with the associated RPC commands. Those commands will be sent through a GUI (which will first ensure that the `allostaticController` doesn't trigger any behavior => all drives frozen).
Here is a first attempt, please add the behaviors you want to demonstrate in it (e.g. Imperial kinematic structure mapping?) .

- Proactive tagging of an object
    - RPC to `proactiveTagging` module: `ExploreUnknownEntity Object Unknown_X`
    - (small) issue: we don't know in advance the exact label of `Unknown_X`, so it should be retrieved directly from the `OPC` or the `sensationManager`
- Proactive tagging of a body part
    - RPC to `proactiveTagging` module: `ExploreUnknownEntity Bodypart Unknown_X`. 
- Goal-oriented behavior from human order with tagging
    - RPC to `ears`: `listen on` (to ensure the adequate grammar is loaded). Should be sufficient unless I forget something. Then the human says *"point to the X"* where `X` is an unknown object.
- **Add your stuff here** ....

jypuigbo, 2016-03-30:
Last [commit](https://github.com/robotology/wysiwyd/commit/e76bbf37549beb904a5747eaf2ae53ed8e4073b0) allows to manually control the drives to: 
- pointing drive
- tagging drive
- narrate
- reset all drives
- freeze all drives
- unfreeze all drives

GUI comming soon!

jypuigbo, 2016-04-01:
You can find a first working version (still not visually appealing) together with this `77e73e677fb4ace6ba34f6064b44662b1a3b3112` and this `363b7020bec1b20776cba29261ea59aacefa1d81` commits. 

Please let us know anything you find or that should be included. 

Normal use would be: 
- (When all `reactiveLayer` is running)
- Press `Manual mode` (everything stops and goes to comfort zone. `AllostaticController` loses control of everything)
- Press some drive you'd like to trigger (e.g. `Pointing`: Every other drive goes to comfort zone and freezes, the selected one goes out and is triggered)
- Press `Autonomous mode` (to give control back to 'AllostaticController' )

Tobias-Fischer, 2016-04-01:
Well done, thanks, nice! We'll give it a try next week :)

/cc @maxime-petit

Tobias-Fischer, 2016-04-08:
Hi @jypuigbo,
Please add the script in the demo XML, along with the `allostatic_plot.py`.

maxime-petit, 2016-04-08:
Freezing drives button is working fine here. Thanks @jypuigbo  @clement-moulin-frier  @sockchinglow. It is very handy during demo to actually have some time to explain :dancers:   We will try the triggering later.

./Max

Tobias-Fischer, 2016-04-08:
Hi @jypuigbo,
This tool is very nice :). Can we add the bodypart moving to it? https://github.com/robotology/wysiwyd/commit/1152bd3835a15593b75c6391635d4c7b26dc6283

jypuigbo, 2016-04-11:
Yes we can. Will be done asap.

Tobias-Fischer, 2016-04-12:
And the kinematic structure please: https://github.com/robotology/wysiwyd/commit/628147ba4ff8429225c87b908a6d4d9338ae5d7f

Tobias-Fischer, 2016-04-16:
@jypuigbo @clement-moulin-frier: Reminder for the two new buttons :)

Tobias-Fischer, 2016-04-19:
And another reminder @jypuigbo @clement-moulin-frier

jypuigbo, 2016-04-19:
@Tobias-Fischer 
With @maxime-petit we discussed the topic and decided that as we could trigger the behaviors by voice, there was no need to add the buttons. Let me know if this still hold. 

Thanks

maxime-petit, 2016-04-27:
Hi,

@jypuigbo : yes, this was mainly for a priority point of view with the demo, it was not needed strictly speaking. But I think it would be a nice features, especially because that mean that we can trigger everything manually.

./Max

jypuigbo, 2016-04-27:
Ok, so I change it to enhancement ;)

Tobias-Fischer, 2016-04-27:
It would be great if you do it sooner rather than later :). I guess it shouldn't take too long.

jypuigbo, 2016-04-27:
@Tobias-Fischer Th thing is: bodypart tagging is not a drive, currently, so it can't be triggered in the same way. Some refactoring might be needed and is already being done, so it should ideally wait till this is finished. 

If you want, what could be done easily is adding a button that sends an RPC command to move each of the bodyparts you have. Leaving the responsibility of knowing if the bodyparts exist to the user. But this ends up being the same as sending an RPC command to the same module. 

Let me know what you prefer.

Tobias-Fischer, 2016-04-27:
Hi @jypuigbo,
Can you please let us know how the refactoring looks like? It would be good to get an early idea what is planned, so @pattacini and others can give some input regarding the establishment of connections between the ports etc. (which is a big issue in the current implementation).

Best,
Tobi

jypuigbo, 2016-04-27:
Hi @Tobias-Fischer, 

Don't worry. I spoke with @pattacini several times during the last two integrations and I am aware of the good practices. I'm also aware of your concerns about the ports, although I disagree about how big is the issue (and we already discussed the topic). I was planning to open an issue regarding this topic as soon as I can, but For this, the needs of the cognitive architecture that drives the behavior must be clear first. I'm sure agreements will be needed and not everything will be perfect for everyone, but I'm sure we all will do our best. 

Patience will be appreciated for the next couple of weeks, but still I'll try to be faster than that. 

Best, 

Jordi

clement-moulin-frier, 2016-04-28:
Hi guys,

Can you  please be more precise about what you are requesting, I don't completely get it. You want to trigger the movement of a random body part? Or of a specific one?

In any case the best way is to send us the RPC command(s) needed to trigger what you want, then creating the button and the callback for it is a matter of minutes.

Clément

Tobias-Fischer, 2016-06-06:
Closing here, we rely on speech recognition instead.

closed 2016-06-06 by Tobias-Fischer