robotology-legacy/wysiwyd #91: Fine tuning robot behaviors

jypuigbo, 2016-03-22:
I'm opening this issue to keep track of the request that was done to UPF during the integration meeting and to keep track of any additional improvement that anyone might like to have / do. 

Specifically, we were requested to make the pointing behavior nicer by looking at the partner when showing one of the objects in the scene. 

We are testing this now and changes are in branch `looking_agent_fromopc`. 

I think that, in order to keep track of such low-impact (hopefully) improvements we could use this issue. 

I hope it makes sense. 

Thanks

pattacini, 2016-03-22:
@jypuigbo I would rather use one issue per development...

jypuigbo, 2016-03-22:
@ugo It's fine with me. I rename the issue accordingly then.

Tobias-Fischer, 2016-03-31:
Fixed with https://github.com/robotology/wysiwyd/commit/e6ee942c49b5ec955b10ccab12559ab7434a875c

jypuigbo, 2016-04-01:
Me an @sockchinglow  found that `agentDetector` doesn't update the `agent->m_present` to `1.0` here in Barcelona, although I guess that it works at least at Imperial. 
Do you have any hint of why this would be? (maybe @maxime-petit ?)

Tobias-Fischer, 2016-04-01:
It's working fine here in London. We've worked on the demo the whole week and didn't have any issues with the agent detector. Are you sure agent detector is updated on the machine you run it on? Is the skeleton shown in the viewer which is defined in the WYSIWYD year 2 demo xml file?

Tobias-Fischer, 2016-04-01:
Also it's not related at all to this issue. Please open another one of the issue persists, closing here.

closed 2016-04-01 by Tobias-Fischer