robotology-legacy/wysiwyd #94: [behaviorManager] Cleanup of duplicated code

Tobias-Fischer, 2016-04-01:
Hi @clement-moulin-frier @jypuigbo @sockchinglow,

@maxime-petit and I discovered that there are THREE different files all concerned pointing in `behaviorManager`. The methods in `followingOrder` and `pointingOrder` are also nearly the same. In fact, I'll make them the same in a minute ..

Please clean some code and remove duplicates, otherwise it's very hard to debug and understand what's going on.

Best,
Tobi and @maxime-petit

clement-moulin-frier, 2016-04-01:
Hi,

Well, `followingOrder` and `pointingOrder` are indeed very close, since `followingOrder` is a generalization of `pointingOrder` allowing to deal with more actions like `push` or `look at`. So `pointingOrder` should be removed indeed. @jypuigbo , can you please confirm?

However `pointing.cpp` is something different. It is about knowledge expression, allowing the robot pointing at known objects by saying there name ("Do you know this is a cube?"). This one has to be kept.

Tobias-Fischer, 2016-04-01:
Happy to keep `pointing.cpp`. However, `pointingOrder.cpp` should be removed. See also the big cleanup of `followingOrder` in https://github.com/robotology/wysiwyd/commit/2ae1423eb2acd0f3f3160a0054032a46110a3585

jypuigbo, 2016-04-01:
@clement is right. We tested the module this week, so now it can be
removed. `Pointing` is a different behavior, that could be renamed into
`knowledgeExpression`

Tobias-Fischer, 2016-04-01:
Cool @jypuigbo. Would be great if you remove it along with all the references to it in other modules, config files etc.

clement-moulin-frier, 2016-04-01:
Thanks a lot for the cleaning, much better indeed :) (please upf look a the mentioned commit :).

I've removed `pointingOrder` in 5bb7c9eadc872473f2c638bdeba48a26e1771796

Tobias-Fischer, 2016-04-01:
See also https://github.com/robotology/wysiwyd/commit/aa685ed24d2f1362c96cd3b6227f11a1605cc9ee

Closing here. Thanks @clement-moulin-frier!

closed 2016-04-01 by Tobias-Fischer