robotology-legacy/wysiwyd #95: [pasar] Pasar is overfilling the ABM

Tobias-Fischer, 2016-04-01:
Hi @gregoire-pointeau,

@maxime-petit and I have discovered that `pasar` sends tens of snapshots to the ABM every second. Especially appearance and disappearance happens very often. Can you please reduce this amount? Otherwise it will fill up our hard drives very quickly .. and make narrative stories a mess as well.

gregoire-pointeau, 2016-04-01:
I never witnessed this behavior.
Can you send me a snapshot of your ABM to check what are the information that go in the ABM please?
Here, we have a snapshot only at appearance or disappearance of an object, which is not very often.

maxime-petit, 2016-04-01:
Actually, the appearance/disappearance is especially when the head is moving (look at the table, look at you, look at the table) which happens quite often in the demo. Because the appearance/disapperance in such case is just that the icub loose track of the object for some seconds, they are not "disappearing" (i.e. they are not removed from the table). 

This thing cannot be replace by a check directly to OPC if an object was present/absent before/after an action? Or making sure the appearance/disappearance is a proper object add/remove to the working space.

./Max

gregoire-pointeau, 2016-04-01:
We discussed a lot this problem at INSERM during the IE.
What you are talking is supposed to be the role of `attentionSelector` which should set the presence to 0 / 0.5 / 1 and `pasar` checks for absence (`0`).
Are you moving the head using `attentionSelector` ?

maxime-petit, 2016-04-01:
No because we agree to not use it for the demo :)

We will try to increase `pasar` option about threshold for appearance/disapperance. By default it is respectively 1 and 2 seconds which in my opinion is way to short (a looking/pointing action take couple of second at least, appearance/disappearance). We will let you know

/Max

gregoire-pointeau, 2016-04-01:
Ok. However, it is strange that you have tens of input per second. Can you send me a snapshot please ?

maxime-petit, 2016-04-01:
Yop!

It is not tens input/second but it definitely have a "flickering" nature as soon as the head is moving (look or home action), as well as some disappearance/appearance separated by a few ms (5651/5652, 5679/5680)

[image]
[image]

gregoire-pointeau, 2016-04-01:
Thanks. It seems that you have between 7s and 10s (at mean) between each apperance and disapearance.
Is it for the same object? I mean I coded a bit fast the thresholds for appereance and dissapearance, but I think they work.

If we decide to increase too much the thresholds, `pasar` won't detect that we bring a new object if we want to do the scenario with different objects.

pattacini, 2016-04-01:
@maxime-petit can you double check if the object is actually disappearing during gaze activity? The tracking system, if working well, should prevent this problem.

Tobias-Fischer, 2016-04-01:
Hi @pattacini,
We've now added some `lookAtAgent()` all over the place, where the objects definitely disappear. It might be the reason. Our robot is having some hardware issues right now, we'll test soon ;)

gregoire-pointeau, 2016-04-07:
Hi @tobias, did you had time to test it?

Tobias-Fischer, 2016-04-07:
Hi @gregoire-pointeau,
Yes it seems to be okay. Thanks :)

closed 2016-04-07 by Tobias-Fischer